arendst/Tasmota #910: spurious operation

tonycwhite, 2017-09-23:
I have 3 sonoff basic, all of which are programmed to the same state, upgraded to the most recent version (5.8.0)
One of my devices seems to spuriously switch on, at random intervals. Looking at the console output, 

`
12:44:02 MQT: Attempting connection...
12:44:03 DNS: Query done. MQTT services found 0
12:44:09 MQT: Connect failed to 192.168.0.10:1883, rc -2. Retry in 10 sec
12:44:09 UPP: Multicast (re)joined
12:44:20 MQT: Attempting connection...
12:44:21 DNS: Query done. MQTT services found 0
12:44:27 MQT: Connect failed to 192.168.0.10:1883, rc -2. Retry in 10 sec
12:44:27 UPP: Multicast (re)joined
12:44:38 MQT: Attempting connection...
12:44:39 DNS: Query done. MQTT services found 0
12:44:39 MQT: Connected
12:44:39 MQT: tele/loungelamp/LWT = Online (retained)
12:44:39 MQT: cmnd/loungelamp/POWER = 
12:44:39 MQT: stat/loungelamp/RESULT = {"POWER":"ON"}
12:44:39 MQT: stat/loungelamp/POWER = ON
12:44:40 UPP: Multicast (re)joined
12:49:09 MQT: stat/loungelamp/RESULT = {"POWER":"OFF"}
12:49:09 MQT: stat/loungelamp/POWER = OFF
`

I have home assistant running elsewhere, with MQTT running. I have set up the switches exactly as per the wiki instructions, and all but one of my devices work happily.

`mqtt:
  broker: localhost
  port: 1883
  client_id: HomeAssistant_1
  username: xxxxx
  password: xxxxxxxx
  birth_message:
    topic: "tele/hass1/LWT"
    payload: "Online"
    qos: 1
    retain: true
  will_message:
    topic: "tele/hass1/LWT"
    payload: "Offline"
    qos: 1
    retain: true
`
`  - platform: mqtt
    name: "Lounge Lamp"
    state_topic: "stat/loungelamp/POWER"
    command_topic: "cmnd/loungelamp/POWER"
    payload_on: "ON"
    payload_off: "OFF"
    optimistic: false
    qos: 0
    #retain: true
`

From the console output, it appears to operate when the mqtt server drops off, then rejoins. 

all devices have unique mqtt client id, as per the user config (dves_last 6 char from mac)
Userconfig was also set to MQTT_POWER_RETAIN = 1 from the wiki.

Does anyone have any clues as to why this happens? I'm guessing its either the device, or the MQTT server causing the issue., but not certain, or knowledgeable enough to dig to deep.

gheesung, 2017-11-26:
I am also having the same problem. I have linked up a Sonoff S20 to a fan and it switches on itself. The following is my console log

`21:32:12 DNS: Initialized
21:32:12 UPP: Multicast (re)joined
21:32:13 MQT: Attempting connection...
21:32:13 MQT: Connected
21:32:13 MQT: brownstone/livingrm/switch02/tele/LWT = Online (retained)
21:32:13 MQT: brownstone/livingrm/switch02/cmnd/POWER = 
21:32:13 MQT: brownstone/livingrm/switch02/stat/RESULT = {"POWER":"ON"}
21:32:13 MQT: brownstone/livingrm/switch02/stat/POWER = ON
21:32:15 UPP: Multicast (re)joined
21:33:45 APP: Serial logging disabled
21:34:16 MQT: brownstone/livingrm/switch02/tele/STATE = {"Time":"2017-11-26T21:34:16", "Uptime":0, "Vcc":3.514, "POWER":"ON", "Wifi":{"AP":1, "SSId":"ngs02", "RSSI":70, "APMac":"78:24:AF:E4:BD:40"}}`

bachoo786, 2018-02-14:
I am having the same issue ever since I set MQTT_POWER_RETAIN = 1

tonycwhite, 2018-02-14:
I put a freshly downloaded software on to the devices, which solved it, but i did think it was something to do with the MQTT retain.

bachoo786, 2018-02-14:
so when I connect a mains cable to the sonoff which is not powered to the mains I notice that I get the same problem as described in the OP. 

I have to do a fresh download and it resolves the issue.

tonycwhite, 2018-02-15:
So the device must be powered somehow

My fresh install was only to remove any settings I messed up.

I have found dry joints on the power connections which also cause spurious
operation!

Tony

bachoo786, 2018-02-15:
does your esp8266 within the sonoff work correctly when powered via the mains?

tonycwhite, 2018-02-15:
The esp had the issue both via mains and powered through the serial
connection.

Tony

bachoo786, 2018-02-15:
I will have to connect it to the mains and see if I face the same problem. I think my issue is to do with the ip address and WiFi connection. Maybe tasmota and esp8266 don't like 5ghz and prefer 2.4ghz band

tonycwhite, 2018-02-15:
I believe esp devices are 2.4ghz only.

Tony

bachoo786, 2018-02-15:
Well mine is connected to 2.4ghz..what have you used for your dns server? Did you use the same address as the address of your gateway?i.e. router ip?