arendst/Tasmota #911: RF Interference?

jobingr, 2017-09-23:
I have multiple Sonoffs running on Tasmota including. 2 normal, 2 dual, power, slampher and 4 pro.

All my non RF enabled Sonoffs operate as expected.   The ones with RF (4CH Pro and Slampher) hardware seem to have random button press events.   I have enabled debug level 4 on web interface and it seems that the API registers this as button presses followed by MQTT signals.

Here is example from console log on 4CH Pro showing issue:

`15:29:24 APP: Button 3 level 1-0
15:29:24 MQT: tasmota/sonoff41/stat/RESULT = {"POWER3":"OFF"}
15:29:24 MQT: tasmota/sonoff41/stat/POWER3 = OFF
15:29:24 APP: Button 3 level 0-1
15:29:24 CFG: Saved to flash at F8, Count 3700, Bytes 1648
15:29:29 WIF: Checking connection...
15:29:29 WIF: Connected
15:29:50 WIF: Checking connection...
15:29:50 WIF: Connected
15:30:03 APP: Button 3 level 1-0
15:30:03 MQT: tasmota/sonoff41/stat/RESULT = {"POWER3":"ON"}
15:30:03 MQT: tasmota/sonoff41/stat/POWER3 = ON
15:30:03 APP: Button 3 level 0-1
15:30:04 CFG: Saved to flash at F7, Count 3701, Bytes 1648`

Button 3 has not been pressed.  Other buttons on this device also come up.    Same happen on Slampher.

I have tried debugging this by pairing these devices with RF remotes.    Real RF signals give similar debug outputs and are reported as real button presses.  My guess is that there are stray signals on RF band causing activation.   I am in Africa and it could be that the bands read from are allocated to some other application in country?    Is it possible to software disable RF commands or further troubleshoot this?

Any suggestions for further debugging from my side?   Does Tasmota have option of bypassing RF signals?

arendst, 2017-09-23:
RF signals are hard wired to either gpio0 (RF) or another microcontroller (4CHpro).

The schematics are in the itead wiki. Perhaps you can cut a wire on the board.

jobingr, 2017-09-24:
Did mod on my 4ch pro.   Pin 11 of U7 syn470r ( the Rf chip) is SHUT pin.  It is connected to ground and internally pulled hight.    I disconnected this pin from ground effectively putting chip in shutdown mode. 

So far so good.   my Open Energy also use 433MHz for communication so I guess that is the source of interference. 

Will give feedback if this totally resolves issue and look at mod on Slampher then

jobingr, 2017-09-24:
Thanks for Tasmota... it is great!

closed 2017-09-24 by jobingr

nmaggioni, 2018-04-16:
Hey @jobingr, did you get around to disabling interactions with the uC? Your tip on U7's pin let me get rid of random interferences, but the module still freezes for a few minutes (as it goes in _learning mode_) if a button is accidentally double-pressed.

jrial, 2019-08-12:
> Will give feedback if this totally resolves issue and look at mod on Slampher then

So, did it?

Reason I ask is we're struggling with the same, and before I attack the thing with the soldering iron, I want to ensure that I'm not lifting that pin in vain.

legalimpurity, 2020-01-20:
FYI DONT RIP OFF THE CHIP (U7), i just did and now the whole device is dead.

e-t-h, 2020-04-06:
I had the same issue. The 4ch pro switched when I sent a 433MHz signal of an other switch. For some reason the 4ch pro had taught in this code. I was able to clear it by using S5 (pressing until green LED flashed). 
To be safe for the future have cut the conducting path coming out from pin 10 (DIN) and short circuited the path to the U8 to ground (at pin 9). If you are careful you can reconnect it later, if you'll need the RF functionality. 

[image]

[image]

HEM59510, 2020-04-10:
Hello, do you think the result is the same if i remove U7 component ?

HEM59510, 2020-04-10:
> I had the same issue. The 4ch pro switched when I sent a 433MHz signal of an other switch. For some reason the 4ch pro had taught in this code. I was able to clear it by using S5 (pressing until [...]

Hello, do you think the result is the same if i remove U7 component ?

e-t-h, 2020-04-10:
I would think that should work also, although I wouldn't solve it that way. If you do so ground the pin 10 to provide a stable input to the STM (U8). (figure it out, the risk is 15 bugs and 6 weeks delivery time..).

jobingr's approach seems to be a very good solution also. I don't have checked if my approach lead to the same issue like nmaggioni described. But I suppose so. I'll check it later and give a statement.

HEM59510, 2020-04-10:
Thanks !

> I would think that should work also, although I wouldn't solve it that way. If you do so ground the pin 10 to provide a stable input to the STM (U8). (figure it out, the risk is 15 bugs and 6 weeks [...]

Thanks, let s us informed. (my need is to disable definitely RF)

Thoky81, 2020-04-13:
Is there any chance to map the S5 button? What GPIO it use? Thanks

HEM59, 2020-08-14:
Hello,
Do you know if jobingr's approach (Pin 11 disconnected) the rolling code process (waiting a RF signal) would be disabled ?
To be more precise, my question, if i push 2 times quickly button channel, my sonnof would still wait a signal ?

iamandymcinnes, 2021-03-28:
Hi, 

Sorry just jumping on an old post, but am I correct in saying if you cut the trace of pin 10 I could just bridge with some solder across the legs of pin 9 and 10?

Also what is the deal with pin 11? Is that worth bothering with and if so would you essentially cut the trace to pin 10 and 11. Then solder bridge pins 9, 10 and 11 together so they are all grounded?

PS this has been driving me mad for ages, I was contemplating replacing them with the new shelly 4pro plus when they come out, but if this works it will save me a small fortune.

Many thanks

Andy

e-t-h, 2021-03-29:
Hi Andy,
it is quiet long ago I struggled with this issue. It is like you wrote: cut the wire (to save the output of U7) and solder a bridge to ground. The U7 sends the detected 433-commands via pin 10 to the following controller. If you cut the wire and connect the input of the following controller to ground there is no more transmission possible. The pin 11 issue I don't know exactly, but I think this is an enable/disable pin for the U7 and will work also. You only need one of these approaches.
The pin 10 approach works to me since I did it. If it would not my father in law had killed me meanwhile, because we control the 6 sprinkler devices in his garden with it ;-). So it will definitivey be a solution for you and will hopefully save you a small fortune..
Ekki

SimoneDomotic, 2021-12-01:
> > Hi Andy, it is quiet long ago I struggled with this issue. It is like you wrote: cut the wire (to save the output of U7) and solder a bridge to ground. The U7 sends the detected 433-commands via pin 10 to the following controller. If you cut the wire and connect the input of the following controller to ground there is no more transmission possible. The pin 11 issue I don't know exactly, but I think this is an enable/disable pin for the U7 and will work also. You only need one of these approaches. The pin 10 approach works to me since I did it. If it would not my father in law had killed me meanwhile, because we control the 6 sprinkler devices in his garden with it ;-). So it will definitivey be a solution for you and will hopefully save you a small fortune.. Ekki
> 
> On the new 4CH PRO r3 (2020-04-13 on PCB) the traces for SYN470R (U3, or U7 on previous revisions) are not exposed above the board. Therefore the only solution left, is to lift pin 11 (shutdown pin if left high).
> 
> **I need to know if the solution of lifting pin 11 also stops the "learning mode" of RF.** If it does not, I will leave original SONOFF firmware and not flash.
> 
> [Please note that in this revision, they also changed the learning mode as being activated when a button is held down for a period of time. This could effect the flashing itself as button S1 (GPI0) needs to be held down upon connecting to flash. It could also effect people who want to connect switches (as opposed to buttons) to the buttons on the board]
> 
> [image]

I have the same problem you solved on the new one it works?

gfn256, 2021-12-01:
Board is still in box! As I did not receive any confirmation that lifting pin 11 helps.

tapd, 2021-12-20:
> I had the same issue. The 4ch pro switched when I sent a 433MHz signal of an other switch. For some reason the 4ch pro had taught in this code. I was able to clear it by using S5 (pressing until [...]

Thanks for that! This worked for me as well...cheers

PRothDC, 2023-05-01:
I had the same issue with two different 4CH Pro R3 devices.  Drove me crazy looking for bad buttons, missing pull up/downs, bad power supplies, etc.  It was definitely a problem with the 433 MHz RF.  My program involved pressing and holding buttons, and when a button was held it would go into pairing mode (unbeknownst to me) and lock onto whatever random signal came along next.

Unsoldering the 433 MHz antenna solved my problem.  Maybe if the RF signals were stronger, that wouldn't have been enough.  Next, I was going to grounding the antenna pad to the nearest ground, probably pin 2 or 3 on the SYN470R , but I didn't need to.

The moral of the story is **_don't use a 4CH PRO if you don't need the RF control_**.  Use the non-pro, which doesn't seem to have any 433 MHz remote control.  Pretty dumb of Sonoff to make it so that holding a button causes pairing IN HARDWARE, and nothing in the firmware can stop it.

Critical-Dmg, 2024-02-10:
I had the same problem, I just installed a sonoff 4ch pro R3 on a rolling Motor Door, and it started going up and down by itself. I figured out that it picks up the signals from car key and turn on all Channels when one button is pressed. Do you think that unsoldering the Antenna will solve it or is there a chip responsible for the RF feature and should remove it all.

PRothDC, 2024-02-10:
Here it is about 10 months after my prior post, and I haven't had any problems after unsoldering the antenna.  If your 433 MHz signals are very strong (like from a very nearby car key fob, then that might not be enough, but it is very easy to do as a first step.  It is the coiled copper wire at to bottom center/right in the picture by gfn256 above; it is about an inch long (if I remember correctly), and it is simple to remove.  Or follow the picture above by tapd if removing the antenna isn't enough.

chrisgrou, 2024-02-12:
Just bought one and I'm having the same problem. Wouldn't pairing their 433Mhz remote solve the issue? Wouldn't it overwrite whatever random signal it picked up by itself?

vladmihociu, 2024-06-28:
> Just bought one and I'm having the same problem. Wouldn't pairing their 433Mhz remote solve the issue? Wouldn't it overwrite whatever random signal it picked up by itself?

Have you tried this?