arendst/Tasmota #913: Counter Sensor Operation

Paulf007, 2017-09-24:
Apologies in advance as I am not sure where I should ask the question. I am attempting to use GPIO14 to measure my pulse prepaid meter with the Counter module. Now the module does "read" the change in the pulse interval as I can see the value change but I am not sure what that value is. 
Where can I get more info on the operation of the counter module?

I am using a LDR with a resistor to pull the voltage down on the LED flash. The voltage drops from 2.6v to 0.9 v so it should be seen as a High Low , I hope . 

Best regards
Paul
South Africa

davidelang, 2017-09-25:
On Sun, 24 Sep 2017, Paulf007 wrote:

> Apologies in advance as I am not sure where I should ask the question. I am 
> attempting to use GPIO14 to measure my pulse prepaid meter with the Counter 
> module. Now the module does "read" the change in the pulse interval as I can 
> see the value change but I am not sure what that value is.

that value should be either the number of pulses since the last report (if it 
reports similar numbers each time) or the total number of pulses since startup 
(if it's reporting an increase by a similar amount each time)

Paulf007, 2017-09-25:
@davidelang thank you for responding! I see that the pulse is on increasing since start up. Am I correct in that if I were to take a value every 5 seconds it should then show the amount of pulses received in the last 5 seconds and from there I could calculate the power consumption? 
(Changing the Telemetry period to 5 )

I will still play a bit more but so far the values that I receive is a constant 50 pulses (That is if I take the previous value and deduct it from the last value received)

closed 2018-05-07 by stale[bot]