arendst/Tasmota #916: Remove retained messages

arun2118, 2017-09-25:
How can i remove the retained messages(LWT) from reappearing on my broker (cloudmqtt)?

davidelang, 2017-09-25:
see the wiki page on poweronstate it includes directions for finding and 
removing such messages

arun2118, 2017-09-25:
Its saying write "mosquitto_sub -p 1883 -u <username> -P <password> -t cmnd/+/power" to check the messages, but where do i do this? console? its asking for Mqtt host username and pass? If the topic i get back is "tele/sonoff3/LWT" i replace power(in the example) with tele?

davidelang, 2017-09-25:
On Sun, 24 Sep 2017, arun2118 wrote:

> Its saying write "mosquitto_sub -p 1883 -u <username> -P <password> -t cmnd/+/power" to check the messages, but where do i do this? console?

at the command line ofthe server your mqtt service is running on (or you would 
have to addapara meter to tell it what host to connect to)

> its asking for Mqtt host username and pass?

ifyou have one set, otherwise ship those parameters

> If the topic i get back is "tele/sonoff3/LWT" i replace power(in the example) with tele?

yes

arun2118, 2017-09-25:
"mosquitto_sub -p 19830 -u ha -P ahat8 -t cmnd/+/tele" returns nothing in cloudmqtt console.  ha being mqtt broker and ahat8 being pass word

closed 2018-05-07 by stale[bot]