arendst/Tasmota #917: "up"grade is a DOWNgrade

hoekbrwr, 2017-09-25:
I tried the update mechanism with the copy at http://sonoff.maddox.co.uk/tasmota/sonoff.bin and find out it does not check if this is a newer version! My existing version was 5.8.0a which I loaded with PlatformIO and after "up"date I have a version 5.8.0.
Why don't you keep the download version in sync with this code source?
Why upgrading when it is a lower or same version???

arendst, 2017-09-25:
That's because you are going back from the development branch to the master branch.

The master branch provides the release version of Tasmota and also the compiled binaries made available by maddox.

The development branch (what's in a name) will always contain newer code which might not even end up in the master version.

hoekbrwr, 2017-09-25:
Oops there I learned something important! I wasn't realising there are two branches.

closed 2017-09-25 by hoekbrwr