arendst/Tasmota #919: Please help with Sonoff touch like/alternative 4 gang wall switch

kevinshane, 2017-09-25:
Hey guys, I've got a new wall switch, it looks like really the same as #495 
but I can't find the exact pins to flash, could someone please help me find them? @ferbulous @anhxco
here are the photos:
[image]
[image]
[image]
[image]

ferbulous, 2017-09-25:
The easier way would be to use OTA method

https://github.com/arendst/Sonoff-Tasmota/wiki/SonOTA---Espressif2Arduino---Tasmota-without-compiling

If you still need to solder, since I couldn't really trace the pins. I just soldered directly to pins on the esp8285 chip

[image]

kruebsam, 2017-09-26:
What is the brand you bought it from? AliExpress? Which one?

kevinshane, 2017-10-01:
@kruebsam hi, please refer to #495 topic, I bought the 4 gang version from @ferbulous's post

kevinshane, 2017-10-01:
@ferbulous hey man, could u also please tell me the PSF-B04 is a 8266 or a 8285? I google it and says the psf-b04 is a 8266 chip, does it means when flashing, I should change to board to a "ESP8266 module" or I should follow the wiki(T1version) to choose a 8285 instead??
Shane

ferbulous, 2017-10-01:
@kevinshane
On the itead wiki, it's already listed as esp8285
Use esp8285 module and flash sonoff 4ch profile

kevinshane, 2017-10-01:
@ferbulous Thank you so much man! It once confused me a lot, thanks !

alton76, 2017-10-01:
Just wondering because I’m getting sync error when flashing PSF-B04
(ESP8285)

To get into flash mode, is it same as ESP 8266 where the GPIO 0 is at low,
connected to ground?

Flash Size: 1M (0 SPIFFS)?

I’m trying to flash the T1 touch, 3gang UK

Forceu, 2018-03-02:
It took me quite a while to flash the image without soldering cables to pins (I definitely don't trust my soldering skills for such a small connection).
The terminal can be used as well, unfortunately I only found out the pins for VCC and GND, but that helped a lot.
To put it into boot mode connect GPIO-0 to GND when booting, when it was in bootmode the LED was not flashing.
Once it was flashed, I shorted GPIO-0 five times to get the board to connect to WPS.

Maybe that helps anyone.

[image]

sabag, 2018-04-09:
hi,
i bought a similar wall switch with 2 gangs.
i would like to flash my own firmware to connect to my own mqtt server. did anyone was successful with flashing ?