arendst/Tasmota #920: Sonoff pow not able tu use temp sensor

RemCom, 2017-09-25:
In de current software versions (5.8) when using a Sonoff pow it is not possible to configure a temp sensor on GPIO1 or GPIO3.

I believe this is still possible. The schematic doesn’t suggest that these ports are used in the energy measuring hardware.

Is it possible to make this available in the software?

arendst, 2017-09-25:
See why not supported  https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-Pow

RemCom, 2017-09-25:
There is nothing on that page where you suggest that it isn’t possible.
The only thing that you mention is that it isn’t a smart idea to connect your pc/laptop to it because you have a change to make a ground loop. But still, just like in the normal sonoffs. The 3.3v is still 3.3v and the ground is always the ground and the serial wires are not used.
If these voltages are doing strange stuff the processor should also fry itself if it would be any different. The only reason why it is a bad idea to connect something to the serial ports is a pc with witch your connecting 2 grounds from a different power supply to each other. That can cause a surge to the hardware because the 2 ground levels are trying to get to the same level potentially frying any hardware in the process.
Al already have a sonoff pow with a local hack running power measurements and having a temp sensor for a few weeks.

RemCom, 2017-09-25:
Sorry for the lack of enters in the text. The got lost somewhere typing the text on my phone

davidelang, 2017-09-25:
actually, the 3.3v is 123v and the ground is 120v

that's why connecting to these pins is not supported. If you hook up an 
opto-isolator you could safely use those pins, but if you can do that, then you 
can modify the sources to support it.

RemCom, 2017-09-27:
didn’t noticed that the ground was a 120 volts. But also without the opto-coplers you can safely use it to hookup a sensor. Just put some shrink sleeve around the pins (which is always a good thing to do). Then its safe to use without a opto copler. Remark that I only talk about connecting temp sensors.

A risk that you still can have is that your temp sensor has a metal case which is connected to the “ground” and having 120 volt on it. That’s not safe. But with a sensor like a AM2302 you have little to no risk.

The problem of keeping my own source for this that I can’t easily update without doing some extra work.

Is it an idea to make this option available but only after you clicked a box that your aware of the danger that your taking? Possibly with a link to the wiki or something like that.

mormor1971, 2017-12-03:
I would agree that it would be great to have the option to add sensors / additional relays to a Sonoff POW as that would open up a lot of options for measuring power to lots of smaller loads without having a Sonoff POW per small load.

Would it not be possible to enable that somehow?

Frogmore42, 2017-12-04:
The code is freely available, just modify it to do what you want.  Sonoff_template.h is the file that you would need to modify.  As others have said this is an unsafe configuration.  While I will work on a live house electrical panel, I chose NOT to add additional sensors to my POW and am using a TH10 and a Wemos D1 for sensors.  But, you are free to do whatever you are comfortable doing.

davidelang, 2017-12-04:
On Sun, 3 Dec 2017, Morten Mortensen wrote:

> I would agree that it would be great to have the option to add sensors / 
> additional relays to a Sonoff POW as that would open up a lot of options for 
> measuring power to lots of smaller loads without having a Sonoff POW per small 
> load.
>
> Would it not be possible to enable that somehow?

The Pow is not supproted becasue on the EU version, there is no low votage, 
'ground' as far as the ESP is concerned is line voltage, so any sensors switches 
you attach are at line voltage, very dangerous.

Anyone who can safely attach things to the Pow can also change the table in the 
code to enable them.

pjanco, 2017-12-13:
The wiki says there is a 50% chance that the GND on POW will be connected to a live AC wire. I have studied the scheme I do not see any reason why this should happend.

Can anyone explain it to me?

The only explanation I've is that someone can use symetric european 2-pin power plug like this:

https://en.wikipedia.org/wiki/Europlug#/media/File:Euro-Flachstecker_2.jpg

But if I use proper connection with 3 separate wires (L,N,PE), it must be a 0% chance that I will have a live AC wire connected to the GND:

https://en.wikipedia.org/wiki/AC_power_plugs_and_sockets#/media/File:CEE_7-7.jpg

I'm right?

PeggyFree, 2018-01-24:
Hello, the wiki says "Also do not connect any additional sensors to serial pins until you are 100% sure."
How to check the real voltage on GND and +3.3V? I have a multimeter.

Frogmore42, 2018-01-27:
The simple answer is one lead of the multimeter (set to AC voltage at a sufficiently high enough range) on ground (from an outlet) and the other on the 3.3v or ground lead from the Sonoff.  But, if you have to ask the question, you probably don't have enough knowledge about the dangers to do it safely.  It is pretty easy to get shocked (which can really hurt) and possible to get killed if the current gets your heart to stop functioning properly.

Frogmore42, 2018-01-27:
Since I was updating an S31 and had the case open, I decided to actually measure the voltage of the gnd that the esp8266 uses.  I had it reasonably safely held in a bench vise and connected it up to the power line with the case open.  I connected my multimeter to the earth ground pin of the S31 (the one connected to the powerline) and the gnd pin on the daughter board that holds the esp8266.

The meter measured **120VAC from earth ground to the gnd pin** (and also from earth ground to 3V3).  This means that any sensor that is attached to any of the esp8266 pins will be sitting at 120 (or 220-240 for people outside USA and Japan) above earth ground.  This can be a lethal problem.  This also matches what I measured on the POW device I set up a few months ago.

Sure, as long as you don't touch it, it is "safe".  There are people who spend their lives determining exactly what one needs to do to actually make something Safe.  I am not one of those people and never even played one on TV.  But, I do know enough to know that I really don't want to learn enough to really do it safely, so it is best to just use the S31 to only control (and measure, in the future) the power.

While I am currently using sensors attached to non-power measuring esp8266 devices which are reasonably safe (they have some isolation from the powerline) I am considering only using esp8266 based devices that are powered by separate low voltage supplies to provide an even safer solution.  Sure, it costs more for the devices and power supplies, but I believe it to be money worth spending to be safer.

However, it is your life (and those that are around you who might touch something accidentally) and you are free to take the risks that you believe to be worthwhile.

lauff, 2018-04-19:
I would need to switch on/off the POW from two meters away. 
I understood that I cannot connect a switch to GPIO.
Would it be safe to "extend" the existing switch by soldering cables and attaching an external switch (or is this also connected to the mains)?

Frogmore42, 2018-04-19:
Yes, everything in the POW can be considered to be at mains voltage and therefore very dangerous.  Get a wemos D1 and have it send a MQTT message to the POW.