arendst/Tasmota #922: Different Topics for multiple Switches

nightcat91, 2017-09-25:
In Version 5.5.1 it was possible to tell which switch was pressed because the MQTT Topic would add a number according to the Switch that was pressed.
This assumes multiple Switches beeing set up, and the "SwitchTopic" beeing set to "sonoff01switch".
Example: 
"cmnd/sonoff01switch/POWER1"  and then it would either send "ON", "OFF" or "TOGGLE" 
(at least the way I use most my Switches, connected to momentary switches, SwitchMode 5)

now if I would hold down Switch 2 it would send "HOLD" to this topic:
"cmnd/sonoff01switch/POWER2"

The next software version I wanted to use was 5.7.1 and I noticed that it does not add the number behind "/POWER" anymore. Now I cant tell which switch was tuned on, off or beeing held down (Hold).
The same is still true for version 5.8.0a.

Did I miss a special setting somewhere?
I would really love to get that small number back, because otherwise I cant use more than 1 Switch per device, since I cannot tell which Switch triggered the publish

davidelang, 2017-09-25:
switches and power only have numbers after them if there is more than one.

If you have two relays, you have power1 and power2, if you only have one relay 
you only have power

same thing for switches.

nightcat91, 2017-09-25:
well... it has been that way for me when I use V5.5.1 but not on V5.7.1 or 5.8.0a

Can you please check this using a real device and configuring multiple switches and the SwitchTopic as I wrote above?
I am using NodeMCU

arendst, 2017-09-26:
@nightcat91 you're right that it used to work that way in v5.5.1 for switches only. I made some changes and removed that possibility by accident. I will reintroduce in next release.

nightcat91, 2017-09-26:
@arendst Thanks for your reply.
I will check it again once I get back home. I am mainly using Wemos D1 mini and NodeMCU units.

nightcat91, 2017-09-26:
Ok, now I was able to do some more tests.
Here is what I did:
- Flash V5.8.0a with almost original "user_config.h" (I only changed the necesarry thing like wifi and mqtt and IP)
- Configure Switch1, Switch2, and Switch3 using the standard "Sonoff Basic" Module
- set switchtopic to a text string of test
- Set SwitchMode1 (and 2 and 3) to SwitchMode 5
Now the results where as expected: the MQTT message is like this:
cmnd/test/POWER1 = TOGGLE
cmnd/test/POWER2 = TOGGLE

- Now I changed the Module to "18 Wemos D1 mini"
And now the MQTT message looses the number behind POWER...
I left everything else the same, I only changed the module to be able to access all the GPIO pins!

Most of the time I use this great software on a NodeMCU or Wemos D1 mini since I use those for switches, sensors and PIR motion sensors around my house. 
And I still have a few devices running right now that are still using the software V5.5.1. Those are configures as a Wemos D1 mini module and they show which Switch was pressed by the number behind the POWER.

I hope this helps, I am sorry it ended up beeing that much text...
Hope you can bring this feature back

arendst, 2017-09-26:
The basic sonoff has at least one relay. I did test it on a wemos with one relay and two switches which worked fine.

I suggest you just add a dummy relay to solve your problem

nightcat91, 2017-09-26:
I can confirm that adding a dummy relay worked for both my test setups (NodeMCU and Wemos D1 mini).
After I used any GPIO Pin and added a dummy relay to the configuration I got the Switch numbers behind the POWER displayed again in the MQTT Topic.

@arendst Thanks a lot for your help. Maybe you can change it some future versions but its not a big deal now that I know how to work around it. Keep up the great work!!

cyberkeiler, 2017-09-27:
I'm missing that feature, too. 
Used a Sonoff Touch with two external Switches. But since by config there are only two GPIOs defined for this module, the amount of dummy relays you can add for a work around is limited, too.
Would love to do this by console again.

arendst, 2017-09-28:
No need to configure dummy relays for a touch; they are already defined by default.

Also, you need just one dummy relay to make the concept of all four switches work.

jchmielniak, 2017-11-18:
The solution with dummy relay on NodeMCU/Wemos works for 'SwitchX'. It doesn't work for 'ButtonX'. Buttons send always POWER without number. Module is configured as Wemos d1 with dummy Relay. Topic set with switchtopic and buttontopic.
Switch:
cmnd/mcu01-switch/POWER1 ON
cmnd/mcu01-switch/POWER1 OFF
cmnd/mcu01-switch/POWER2 OFF
cmnd/mcu01-switch/POWER2 ON

Button:
cmnd/mcu01-button/POWER TOGGLE
cmnd/mcu01-button/POWER TOGGLE

davidelang, 2017-11-18:
if you have more than one button defined, it will include a number. If you only 
have one switch defined, it will not

backwards compatibility to the time before multiple buttons/switches/relays were 
supported

closed 2018-05-07 by stale[bot]