arendst/Tasmota #926: No NTP update with static IPAddress

gabbermorph, 2017-09-26:
When i have set a static IPAddress in my Sonoff Basics. 
Then i don't get NTP time updates, it stays at 1970 something.

Cheers

arendst, 2017-09-26:
Without a static dns ip address the ntp servers cannot be reached.

jetema, 2017-09-26:
Hi gabbermorph, I have several sonoffs all of them with static IPAddress and NTP servers update ( after a few seconds ) right, here's a image :
[image]
[image]
Probable you must have some configuration wrong

gabbermorph, 2017-09-27:
Okay, thats intresting.
Why don't you use the Gateway as your dns server? I used also google dns and it won't work.
Maybe i need to fiddle around with that.

Cheers

arendst, 2017-09-28:
If your Google DNS doesn't work you probably did not enter the gateway address too.

For Static IP addresses you will need to provide all four IPAddress parameters as documented in the wiki > Commands

rlust, 2017-09-28:
Still having issues with getting NTP to update also. I have tried google DNS and and local. Works for others but not some with same settings. Not sure why it comes and goes.

gabbermorph, 2017-10-05:
do u have the problem even with DHCP enabled?

garywilson4, 2018-02-10:
none of my sonoffs update the time from the ntp servers no idea why any help appreciated