arendst/Tasmota #928: How to use api/upload-arduino.php from PlatformIO

lgnd33, 2017-09-27:
Hi, 

I see that there's a helper script that should, judging by the code, upload compiled binaries to OTA server.
I recon it somehow works in Arduino environment, but wonder if anyone has equivalent functionality for Atom/PlatformIO?

arendst, 2017-09-28:
FYI file upload-arduino.php is stored on the webserver. In the arduino environment the file espupload.py is used.

I saw your PR using SSH. Perhaps you can tweek espupload.py to use the upload-arduino.php webserver environment.
At least you gave me a starting to point to look into it too...

lgnd33, 2017-09-28:
I saw espupload.py and saw it's made for Arduino environment, which wouldn't work (directly) as PIO uploader.

As I only need scp, I made a simple redefinition of PIO uploader to use scp instead of esptool (normally used to upload using FTDI), but I imagine folks interested in other methods of upload (HTTP/HTTPS, FTP, mv, rsync, git, you name it) would be able to change the commands and use something else. 

In the end, if there's a lot of interest, we can maybe create build-environment-agnostic upload script, that would have light triggers from actual build environment.
For now, simple scp from PIO works for me, hence the PR :)

arendst, 2017-09-30:
Based on your information I managed to get the standard Arduino OTA upload file as provided in the arduino folder to work with pio.

Steps taken:
1 - make a folder ``pio`` in the root of the pio source folder as you did.
2 - copy file ``arduino\espuploader.py`` to ``pio\espuploader.py``
3 - make a file ``pio\http-uploader.py`` with the following contents:
```
Import("env")
env.Replace(UPLOADCMD="pio\espupload.py -f $SOURCES")
```
4 - Add in file ``platform.ini`` for every [env:sonoff...] the following key/value pair
```
extra_scripts = pio/http-uploader.py
```
Now compile and upload your source and the binary will be send to the OTA server as was possible with the Arduino IDE.

Note: The copying of file ``espupload.py`` is not really necessary but it provides a clean separation for future use.

closed 2017-10-03 by lgnd33