arendst/Tasmota #930: USE_DOMOTICZ disabling

alpacatastic, 2017-09-27:
Hi loving Tasmota!  Have noticed one very minor 'issue' in user_config.h where it says to disabled Domoticz by putting // beforehand.  I did this and also commented out the following 3 Domoticz lines but then compile fails because the line 

sysCfg.domoticz_update_timer = DOMOTICZ_UPDATE_TIMER;

in settings file needs that value from user_config.h.  If I just comment out the first USE_DOMOTICZ line it is fine so as I say not a major issue but first instinct was to comment all DOMOTICZ lines out.

arendst, 2017-09-28:
Disabling USE_DOMOTICZ only will get rid of all Domoticz support.

arendst, 2017-10-30:
Implemented in master release v5.9.0

closed 2017-10-30 by arendst