arendst/Tasmota #932: Sonoff RF; help me decode acceptable sonoff 433MHz transmission pulse sequences

kessack, 2017-09-28:
I am trying to pair my remote with the sonoff rf which has the custom firmware.

I have a 433MHz standard "garage" remote, identical to the sonoff supplied remote but my remote does not have an acceptable sonoff code sequence on it, so the sonoff rejects learning this code when I ask it to suck it up. 

If I had the structure of the right code I could re-program the remote.
 
I know the sonoff won't learn the code I give it from the remote because when I press the sonoff on board button twice quickly, so that it flashes red indicating it is attempting to suck up the remote code, and press the 433MHz remote button to send the code at the same time, nothing happens. 
The sonof does not blink red 3 times to confirm it has sucked up or learned the code.

The remote I am using is programmed with 24-bits serial; a mixture of high-to-low or low-to-high each forming a 1 or 0 serial radio signal.

I think the sonoff is looking for a particular starting code sequence, or perhaps tri-state, or some other sequence type, which is not how my remote is currently programmed.

I have a bag full of these remotes from aliexpress and spending on a sonoff one seems rediculous!

If someone has a sonoff remote can you sniff the code and publish the code sequence.

That way I can send it from a software signal generator on my laptop to my 433Mhz transmitter, suck it up onto my remote, then program the sonoff with the remote. 

I will then by trial and error publish a number of the sonoff 433MHz remote code sequences that will help other people get going.

jetema, 2017-09-28:
Altought Tx frequency is 433,92 Mhz some garage remotes use their proper encoder ( MM53200, UM3750, PT2262 specially old ones as your remote ( 1 - 0  switches or 1 - 0 - Z ), so is not ensured sonoff will be able to learn your code, try others garage remotes if available
Have a look at this page 
http://lmpautomation.com/blog/?p=24

ionciubotaru, 2017-09-30:
I suggest a better way:
your garage door has a pair of contacts for external switch
Connect them to a relay (no electric potential on the relay)  like this http://www.electrodragon.com/product/wifi-iot-relay-board-spdt-based-esp8266/ and close this circuit via mqtt commands. In this simple way you can stlli use your RT remote control.

lhcarsten, 2017-10-05:
> If someone has a sonoff remote can you sniff the code and publish the code sequence.

I recently ordered such a remote for exactly that purpose. I have an RFToy [https://rayshobby.net/](url) that should do the job. As soon as I have the sequences I will post them here. It should arrive latest 20 oct 2017.

I would like to be able to use the RF function too and have been looking for answers. I am quite new to this so there are still a lot of unknown unkowns on my part. One thing I seem to have found is that if I use the pairing function of the button on the RF receiver itself, I can get one step further. You need to open the Sonoff box to access it. Here is info on the module: [https://www.itead.cc/wiki/433Mhz_RF_receiver](url). It looks as if I am able to pair a remote this way. After pairing, when I press my remote, the LED flashes indicating that a signal is being received. However I haven't figured out how to get this to toggle the relay. Information from Itead seems to suggest it should be possible, but I am as yet unclear if the firmware needs to accomodate this option.

From [https://www.itead.cc/wiki/433Mhz_RF_receiver](url):
Usage
- Pairing: press SW button for 2s, indicator led turns on to enter into pairing mode; release the button, indicator led turns off. Press one of the buttons on RF remote(one button every time), indicator led flashes 4 times every 300ms, which means paired successfully, saved value of the button and exited pairing mode.
- Output: press a successfully paired button to receive valid data, indicator led will be on for 200ms, data port (corresponding one in D0~D3) will be leveled up for 200ms.
- Clear: press SW for 8s, indicator led will be on at 2s and off at 8s. All pairing codes will be cleared, all buttons will be invalid

lhcarsten, 2017-10-14:
OK I got my remote today and did some sniffing. For more info on the RFToy see [here](https://github.com/rayshobby/RFToy) and [here](https://rayshobby.net/cart/rft).

( Bear in mind my knowledge level on this is quite limited. I am just starting to get into this.)

So the codes I got with my RFtoy are:

(Button A) BF88B1------01A5
(Button B) BF88B2------01A5
(Button C) BF88B4------01A5
(Button D) BF88B8------01A5

Since this is a toggle button the dashes in this list stand for the missing other button.
An actual code being sent would be (button A): BF88B101A5

I tried a few other remotes I have. Some of those codes I was able to get to work, but it seemed a hit and miss thing. Pressing the button on the remote a few times seemed to help. From the little sampling I did, it seems that of the last 4 digits the 01 is necessary. Some codes with a D in the position of the A worked. When I sniffed the original remote, the last digit sometimes read 6, 7 or 8 (!?), but mostly it was 5. The first part (first 6 digits) can vary it seems, maybe considerably. A code I got working was: FFFF3001CF

In the spirit of collecting information I tried most of this on an unmodified Sonoff RF. I discovered that programming the radio receiver by using the button on the receiver itself, did seem to record the code, but it wouldn't switch the relay. With the ESPurna firmware I was able to program the Sonoff RF using the quick 2-button press. With Tasmota I haven't yet been able to program the radio receiver in a way that toggles the relay. I suspect this has to do with the [button usage](https://github.com/arendst/Sonoff-Tasmota/wiki/Button-usage) in Tasmota.

One last note. The radio receiver seems to have a memory for several codes. But only the last successfully recorded code is able to switch the relay. And after recording several codes the memory won't accept anymore (I think), and the memory needs to be cleared (quick 3-button press).

stronkje, 2017-12-16:
I am still struggling to program RF codes on my sonoff RF, also with (now) ESPruna firmware.
I want to use my Domoticz with RFlink attached to send RF codes to the sonoff. I know I can program RF codes in Domoticz, via Setup->hardware->Create RFlink devices. There I can input an RF code, like ◦	20;14;FA500;ID=BF88B101A5;SWITCH=01;CMD=ON. 
I double click to put the sonoff in learn mode, but it refuses to learn the code, at least it does not switch the relais, and no state change is reflected in the MQTT messages I see in putty.
Does anyone know how the RF-code should look like, that I can program in DOmoticz?

lhcarsten, 2017-12-19:
Well if it won't learn the code it clearly cannot respond to it. Maybe you just need to clear the memory of the receiver (quick three button press). 

I have a sonoff RF that is not functioning properly. It will learn the code using the button on the receiver itself but will nevertheless not switch the relay Also a normal button press does not switch the relay. My understanding is that the receiver simulates a button press, so that could explains why the relay won't switch. However I can still use the button to get the sonoff into programming mode, so it is not a simple break in the connection of the button itself. And it will still work as a regular sonoff, reacting to mqtt commands.

Two other sonoff RFs both work with the remote. One is running the ESPurna firmware, the other is still running the original firmware. The one running ESPurna will learn and use the code either using the regular two button press, or using the button on the receiver itself (2 second press). The one with the original software will only switch the relay if the code was learnt using the quick 2 button press.

In all cases I get the regular expected feedback through the red led. 

Don't forget you need to bridge R21 to be able to program the Sonoff RF, but to program the receiver R21 should NOT be bridged.

Last thing I noticed is that a short press on the remote doesn't switch the sonoff. I generally have to press and hold the button on the remote a full second.