arendst/Tasmota #933: Request : make sonoffs talking together easily

PeggyFree, 2017-09-28:
Because my sonoffs are now managing lights and sometimes heating, it is critical to get them working even if the mqtt or home automation service is down.

I'd like to see a direct HTTP command feature : 

- If switchx status change to ON, send one or more HTTP requests. (for instance http://x.x.x.x//cm?cmnd=Power%20On)
- If switchx status change to OFF, send one or more HTTP requests. 

With this feature, sonoffs can work together with minimal response time and very low failure probability.

lgnd33, 2017-09-28:
Hi, 

How about investing in highly available MQTT installation? 

There's a number of tutorials out there that explain how to setup a Mosquitto in HA mode and all you need is two R-PIs.

davidelang, 2017-09-28:
Implementing something like this would be a lot of work and eat up a lot of 
program space (which is very tight).

A better approach is to make your MQTT server redundant, or make it run on your 
wireless access point (if your AP is down, your http messages won't get anywhere 
anyway)

PeggyFree, 2017-09-29:
My mqtt server + HA system is already on a virtualized machine un order to
make its recovery easier.  mqtt is not the critical point, the whole ha
system must be working for my Sonoff lights to trigger. I'll Google what
Lazar call "Mosquitto in HA mode and all you need is two R-PIs".

It would have been nice to trigger at least one sonoff with another, even
something ultra simple. But if space is too limited, well...

closed 2017-10-01 by PeggyFree