arendst/Tasmota #936: Sliding motion on WS2812 clock

Amli2011, 2017-09-29:
The hour hand of the WS2812 ledstrip in clock mode jumps from one hour direct to next, but on every analog clock it will slide, bit for bit trough the whole hour.

To mimic that behaviour and to make it more readable as an analog clock i propose the following change in file:
xdrv_ws2812.ino
line 138:
  i = (rtcTime.Hour % 12) * (50 / clksize);
changed to:
  i = (rtcTime.Hour % 12 * 5 + rtcTime.Minute / 12 ) * 10 / clksize;

The changed line will take the minutes into account to make the hour hand representation move with 60 positions instead of 12.

After this fix 7:59 will look more like 08:00, before fix 7:59 looks more like 7:00. 

PS
I am not used to github, could probably issued a pull requst or something, but as your software is near perfect this probably is the only programming contribution from me.

Thank you!

arendst, 2017-09-30:
Ah, it was supposed to be a gimmick but nice to know someone is actually using it. 

I'll make the proposed change in next release.

wunderbaum, 2017-10-01:
Just curious, what is your hardware Setup?

Amli2011, 2017-10-01:
@arendst: Thank you! Both for the software and the response!

@wonderbaum: at the moment it is a ESP-01 module, a 3.3 ldo regulator, a resistor and an diode for level convertion of the digital signal.
This is driving a led strip with 60 ws2812, and is powered by a 5v psu.
And now that the software is solved it is time for the hardware, it is going to be something round in plexi glas ( polycarbonate, lexan or whatever it is named) either plexi painted white with numbers without paint and the leds shining outward behind the plexi and trough the numbers  or as the second idea, as edge light on the plexi with numbers carved in.

It is going in my bedroom, so it will be faded together with light dimmer on main and bedside light via mqtt. I am also looking into giving some warnings and info by flashing the whole clock in different colors.

closed 2017-10-01 by Amli2011

TommySharpNZ, 2017-11-11:
Cool, make sure you share some photos somewhere!

Amli2011, 2017-11-18:
I have begun to document it over at hackaday.io:
[https://hackaday.io/project/28194-esp-and-ws2812-based-clock](
https://hackaday.io/project/28194-esp-and-ws2812-based-clock)

I'am not good or fast at doing documentation, but I will get enough up for you to replicate it.