arendst/Tasmota #937: PowerOnState problem

gabbermorph, 2017-09-29:
Hi, 

i am using an Sonoff Basic with PowerOnState 2. So i can Toggle it on and off by the light Wallswitch.
I am using also Home Assistant with MQTT and hue emulation for Alexa using.

This is the Debug Log after toggle ON it switches off after a view seconds. It also says first i am OFF, but it is on for a view seconds.

> 00:00:00 CFG: Loaded from flash at F9, Count 241
00:00:00 APP: Boot Count 73
00:00:00 Project sonoff Sonoff02 (Topic sonoff02, Fallback DVES_E7C5D3, GroupTopic sonoffs) Version 5.8.0a
00:00:00 CFG: Saved to flash at F8, Count 242, Bytes 1648
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Patch issue 2186
00:00:00 WIF: Connecting to AP1 TiJay in mode 11N as sonoff02...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:03 WIF: Checking connection...
00:00:03 WIF: Attempting connection...
00:00:04 WIF: Checking connection...
00:00:04 WIF: Connected
00:00:04 DNS: Initialized
00:00:04 HTP: Web server active on sonoff02.local with IP address 192.168.177.45
00:00:05 APP: (UTC) Fri Sep 29 07:39:17 2017
00:00:05 APP: (DST) Sun Mar 26 02:00:00 2017
00:00:05 APP: (STD) Sun Oct 29 03:00:00 2017
09:39:18 MQT: Attempting connection...
09:39:19 DNS: Query done. MQTT services found 0
09:39:19 MQT: Connected
09:39:19 MQT: tele/sonoff02/LWT = Online (retained)
09:39:19 MQT: cmnd/sonoff02/POWER = 
09:39:19 RSL: Received Topic cmnd/sonoff02/power, Data Size 3, Data OFF
09:39:19 RSL: Group 0, Index 1, Command POWER, Data OFF
09:39:19 MQT: stat/sonoff02/RESULT = {"POWER":"OFF"}
09:39:19 MQT: stat/sonoff02/POWER = OFF
09:39:20 MQT: tele/sonoff02/INFO1 = {"Module":"Sonoff Basic", "Version":"5.8.0a", "FallbackTopic":"DVES_E7C5D3", "GroupTopic":"sonoffs"}
09:39:20 MQT: tele/sonoff02/INFO2 = {"WebServerMode":"Admin", "Hostname":"sonoff02", "IPAddress":"192.168.177.45"}
09:39:20 MQT: tele/sonoff02/INFO3 = {"RestartReason":"Power on"}
09:39:21 CFG: Saved to flash at F7, Count 243, Bytes 1648
09:39:21 MQT: stat/sonoff02/RESULT = {"POWER":"OFF"}
09:39:22 MQT: stat/sonoff02/POWER = OFF
09:39:29 MQT: tele/sonoff02/STATE = {"Time":"2017-09-29T09:39:29", "Uptime":0, "Vcc":3.208, "POWER":"OFF", "Wifi":{"AP":1, "SSId":"XXXX", "RSSI":40, "APMac":"XXXXXXXXXXX"}}

davidelang, 2017-09-30:
09:39:19 RSL: Received Topic cmnd/sonoff02/power, Data Size 3, Data OFF

This line is the device being sent he command to turn off

So it's very possible that the device was turned on by the button on the device 
prior to the power outage, but that the last MQTT command it got was to turn 
off and the mqtt server retained that message

you really should not mix poweronstate remembering the last setting with mqtt 
retained messages. It just gets messy.

gabbermorph, 2017-10-01:
I did a small workaround. I set sonoffs in home assistant to retain : false. And set retain in the sonoff to on. Now I can toggle the Relais with the wall switch. 
Why can’t the sonoff send a mqtt stat when it’s toggled on?

Cheers

davidelang, 2017-10-01:
it does send a stat when it's toggled, but what is retained and replayed to 
change it's state is a cmnd/ message, and it can't send that when it's toggled 
or you have a loop that will toggleit again when it gets the cmnd message 
back from the mqtt server.

tefinger, 2017-10-20:
> I did a small workaround. I set sonoffs in home assistant to retain : false
Thanks this helped me to solve the problem I had with openHab2 👍 
My Sonoff Basic (with PowerOnState set to 1) turned off a few seconds after power on because the messages were retained in mosquitto. After disabling retaining of messages and deleting all retained messages it now works as expected!

closed 2018-05-07 by stale[bot]