arendst/Tasmota #938: [S20] both LEDs are turned on/off equally

CWempe, 2017-09-29:
The S20 has two LEDs behind the button.
green and blue.

Both tun on/off at the same time, which does not make much sense to me.

If possible these LEDs show reflect different states.
Since the blue one seems to be related to the relay (https://github.com/arendst/Sonoff-Tasmota/issues/488), I think the green one should reflect something like "S20 is connected" or "S20 has Power".
So the green LED should light up always or never.

Is this (already?) possible?
Does this make sense?

flexiti, 2017-09-29:
green one connected to GPIO13 , blue to GPIO12 with realy in parallel

arendst, 2017-09-30:
Take a look at command ledstate to change the green leds functionality.

CWempe, 2017-10-15:
Thanks.
`LedState 0` was what I was looking for.

closed 2017-10-15 by CWempe