arendst/Tasmota #939: IR Receiver?

lgnd33, 2017-09-29:
Does it make sense to implement an IR Receiver?
 
I have a board that has one soldered and ready at GPIO04 (board is "MagicHome LED strip controller"... has a IR variant that comes with cheap IR remote, but codes are really easy to pick up using already included IRremote library).

I don't need it (at least not for now ;) so I can just ignore it, but I figured someone might have a use-case for picking up IR codes and sending them over MQTT - maybe your LED strip controller is close to the TV and you'd like to pick up that "switch source - CATV" on TV remote to dim the lights or something... 

I wouldn't go as far as to implement IR code interpreter in Sonoff-Tasmota - I'd just pick whatever IR code shows up on the IRreceiver and send it off to MQTT for actual interpretation / processing. 

@arendst - it's a nice exercise for me, but don't want to waste space on code nobody is going to use. What's your take?

davidelang, 2017-09-30:
I would like to see that, it seems like a very useful feature to complement the 
IR transmitter.

arendst, 2017-09-30:
@lgnd33 Lazar nice idea and if you manage to keep it's code footprint small it could be integrated with Tasmota. As you stated correctly let the interpretation and processing being done elsewhere.

Give it a try!

lgnd33, 2017-09-30:
Just to check - new GPIO functionality is added to the end of the upins_t, not to cause backward compatibility issues?

arendst, 2017-09-30:
Correct.

closed 2017-10-03 by lgnd33

gnkarn, 2018-01-29:
im , very interested on this , as i need to sense codes from my alarm system, and the IRreceive code is perfect for this ( even  when the code is not coming from IR) .Would like to know if there was any progress , before starting my own branch , tks