arendst/Tasmota #941: H801 support?

ageorgios, 2017-09-30:
Are there any plans for H801 support?

https://github.com/diogos88/HA-H801-ESP8266-MQTTJson-RGB

arendst, 2017-10-01:
Will be in next (pre)release. 

It will also disable direct PWM control for H801 and MagicHome as it is then under control of commands Color and Dimmer.

nosaj66au, 2017-10-01:
Hi Arends,
   Would be great if we could name or title the channels and not have them a just RGB,W1,W2. I'm using one H801 per room giving me dimming control of 5 (24vdc) LED downlights. I using a spare GPIO as the switch like a normal sonoff. This either turns the LEDs off / Returns to last dimmer or went held 5 sec brings LEDs to full brightness. I guess there could be a heap of other options.
What do you think?
Jason

lgnd33, 2017-10-01:
I started working on similar code - are you refactoring xdrv_sfl? 
Can you share the idea, maybe we can split?

//_

arendst, 2017-10-01:
I think the PWM code is complete as implemented in xdrv_snfled.ino. 

Your IR code could go into xdrv_ir_send.ino. If need be I will transfer to other parts if you have something working.

closed 2017-10-01 by ageorgios

lgnd33, 2017-10-01:
I was referring to adding color handling to MagicHome module and
refactoring xdrv_snfled to allow more than 2 direct PWM channels.

Looked at that - seems current code depends on the snf_flg to determine
number of channels and assumes that 1 or 2 channels are direct PWM, while 4
and 5 channels are controller via My92x1.
I was thinking of changing the code to:
- keep 5 component color code independent of the hardware implementation
and use functions like dim, animate (transition) and rgb2hsb on it.
- calculate PWM out of generic color model.
- have a separate per-board configuration for determining controller method
(direct PWM, WS2812, My92x1 or maybe something else if they come up with it)

If you're shooting in a same direction, maybe we can split the work...

arendst, 2017-10-01:
It's already there in the version released today. Just select MagicHome and you have full color control.

In fact any pwm controlled device can now be controlled that way once you enable this feature with command SetOption15 1

Further action will make the code too big while adding minor progression.

See releasenotes regarding correct pwm gpio selection.

lgnd33, 2017-10-01:
OK, just pulled latest version and see that you expanded the meaning of the
snf_flg, which solves some issues.
Let me check the rest of the code and see if anything's missing.