arendst/Tasmota #943: Feature request: RfReceived to Domoticz

mitkodotcom, 2017-10-01:
As the RF bridge actually decodes all received events from all compatible devices (not only paired ones), it would be very handy to feed those events to Domoticz for further processing. This way the bridge can be used to receive the signal from all sorts of door/window sensors, motion detectors, smoke and CO detectors, etc.

My temporary solution is so simple, that it does not deserve a Pull request:

I have added

domoticz_sensor6(rid);

after

mqtt_publish_topic_P(6, PSTR(D_RFRECEIVED), svalue);

in xdrv_snfbridge.ino

This way I am reusing the 6-th sensor index - the only 32-bit one. Now the RF bridge sends the received codes (all of them) as **decimal** numbers to the specified Domoticz sensor (Sensor idx 6 - Count).

In Domoticz I have created a virtual sensor of type Text. I choose the text type just in case: if in the future the code is to be sent in hexadecimal. Currently it will work with almost any digital type (counter, etc.). The following Lua event script then controls the door contact:

for deviceName,deviceValue in pairs(otherdevices) do
    if (deviceName=='RF received') then
        if deviceValue == "11844874" then
            commandArray['Door'] = "On"
        elseif deviceValue == "11844878" then
            commandArray['Door'] = "Off"
        end
    end
end

By adding just one line of code, I have a very cheap and almost universal RF receiver. The above example works with the following door sensor, which sends two different codes - B4BD0A for open and B4BD0E for close: https://www.aliexpress.com/item/433MHz-Wireless-Magnetic-Door-Sensor-Door-Detector-Door-Contact-Detect-Door-Close-Open-for-G90B-WIFI/32805891010.html

arendst, 2017-10-01:
How ingenious! I love one-liners ;-)

arendst, 2017-10-30:
Implemented in master release v5.9.0

closed 2017-10-30 by arendst