arendst/Tasmota #945: Light not turned on when dim level is set

emontnemery, 2017-10-01:
According to https://www.domoticz.com/wiki/Domoticz_API/JSON_URL%27s#Set_a_dimmable_light_to_a_certain_level, the light should be turned on when dim level is set.
However, there seems to be a bug on this line in domoticz_mqttData():
`if (sfl_flg && (sysCfg.led_dimmer[i] == nvalue)) {`
If dim level is unchanged, light will not be turned on.

Maybe the above line should be changed to:
`if (sfl_flg && (sysCfg.led_dimmer[i] == nvalue) && ((power >> i) &1)) {`

arendst, 2017-10-02:
Verified. Will fix in next release

emontnemery, 2017-10-02:
I saw the comment in #946.
Why do we need to send a "Set Level" message to Domoticz when the switch is being turned off?

emontnemery, 2017-10-08:
Here's a new attempt: Just send dimlevel together with the on/off status message. One concern is that Domoticz seems to ignore dimlevel sent with an "off" status message, I'm not sure if that will/might have side effects?
(I did mess up a bit when rebasing tha pull request, sorry about that)

closed 2017-10-11 by emontnemery