arendst/Tasmota #947: Unresponsive button when MQTT connection fails

boyd91, 2017-10-01:
When the connection to the MQTT server fails the button on the Sonoff Touch / S20 becomes unresponsive. The buttons still works -sometimes- but starts to 'hang' regularly. This is probably because it's caught in the loop of trying to reach the MQTT server and therefor cannot handle the button press.

It would be very convenient if Tasmota could deal with a broken MQTT connection better. So that when there's is a problem with the network the physical Sonoff Touch button still works properly.

It removes a single point of failure and keeps me from having to explain to my girlfriend that she cannot switch the lights because there's a problem with my MQTT server 😅

davidelang, 2017-10-01:
There is a bug in the TCP library that tasmota uses, when it is attempting to 
establish a new connection, everything stops for up to 5 seconds.

ionciubotaru, 2017-10-02:
Please check the latest mqtt library @arendst update it few days ago.

emontnemery, 2017-10-07:
@davidelang, do you have any info about the bug in the TCP library?

davidelang, 2017-10-07:
just that during the timeout while it trying to connect, it is waiting, so 
nothing else can take place during that time.

They are 'not waiting correctly', that is, they are not releasing the CPU so 
that other things can run during this time. This is probably not a trivial fix 
because there is probably a lot of software out there now that assumes that the 
other stuff won't run until you have a connection. Tasmota has a lot of stuff 
that's interrupt driven (button presses, timers, etc), and the main loop that 
we want to have continue to run while the connection is being established, so 
they probably are going to need to create a new function that starts the 
connection, and has a callback to execute when they get the connection, and 
another callback to execute if the connection fails

orso7, 2017-10-23:
any progress on this? My girlfriend is driving me nuts because the switch is sometimes unresponsive. Is the mqtt server the problem? I am using Alexa to control the switch.
would a separate mqtt server (RPi with wired connection) help in any way?

arendst, 2017-10-23:
In no particular order:
1) Get a new girlfriend
2) Disable MQTT
3) Use command ``MqttRetry`` to increase retry time

orso7, 2017-10-23:
1. have to think about that, there are other advantages she has....
2. is that possible over the console? can't find the option in the webinterface
3. increasing the retry time will still introduce a delay of several seconds on the switch if the server does not answer.

arendst, 2017-10-23:
Check the Other Configuration tab:

[image]

davidelang, 2017-10-23:
This is not something the tasmota project can solve, it's a bug in the Arduino 
TCP library. While it is attempting to establish a connection, nothing else on 
the device will work.

If you are connecting to a cloud MQTT server, switching to a local one on your 
network would significantly increase your reliability.

orso7, 2017-10-23:
Wow, could have seen that myself!... Thanks a lot, seems like it works now.

closed 2018-05-07 by stale[bot]