arendst/Tasmota #949: Sleep command

Gary202, 2017-10-01:
I'm using a standard ESP12 running Tasmota that's battery powered and the sleep is not sleeping. i have it set too 100 and its not sleeping and the battery will only last around 1 or 2 days. im useing a 9v battery but soon going to be upgradeing to a 1500mah lithium battery. should gpio 16 be tied to reset for the sleep to work like deepsleep?

ionciubotaru, 2017-10-02:
Running esp8266 in deep sleep needs a different software. All the code must be run in setup section, not in loop. In few words you must write your own code.

joba-1, 2017-10-02:
No, ionciubotaru, esp8266 can enter deep sleep at any time you like. Just the simple examples provided with the sdk/core use that pattern to not leave setup. 

As for tasmota: I dont know how or if it works. For deep sleep you definitely need gpio16 to connect to reset.

arendst, 2017-10-02:
Deep Sleep is not supported by Tasmota.

stefanbode, 2017-10-07:
@Gary202 I have a fork of the TASMOTO that I keep up to date and have some additional features, that matters if you want to get beyond SONOFF devices. Take a look at my user here and you should find the fork. It has currently 5.8.0b version. I also introduce and ULTRADEEPSLEEP, because the ESP cannot sleep much longer than one hour. You don't have to configure something. You can just use deepsleep 7200 or higher for example. The trick is to use rtcMem and wake up for less than 0.2 seconds every hour. Works like a charme. I have many battery driven WEMOS or pure ESP8266. In general I prefer to use standalone ESP8266, because they need much less battery than the board versions like WEMOS.

davidelang, 2017-10-07:
On Sat, 7 Oct 2017, stefanbode wrote:

> @Gary202 I have a fork of the TASMOTO that I keep up to date and have some additional features, that matters if you want to get beyond SONOFF devices. Take a look at my user here and you should find [...]

so what is the link to the fork? and what are the additional features (besides 
the long sleep)?

how do you respond to button presses and mqtt messages that are sent while you 
are asleep?

stefanbode, 2017-10-09:
Hi @davidelang,

the link is here: https://github.com/stefanbode/Sonoff-Tasmota/tree/master/sonoff

Additional features are 
- DeepSleep support up to unlimited (deepsleep=xxx [sec]), support also 1day or more deepsleep
- Improved Startuptime incl TLS MQTT and measurement in <9 sek
- Support for Ultrasonic distance measurement HC-SR04
- Support for I2C 10bit 4-channel A/D converter ADS1115
- Support for I2C 8-channel DIO extension board PCF-8574 (large extension to support 32 relays and pulsetimers on 32 relays; other functions still stick to 4)
- Report total UPTIME (seconds) between two deepsleep cycle. This allows to see uptime, even if connect was not successfull or there was a reboot. Using RTC memory to ensure correct uptime
- Support UtraDeepSleep on any duration in seconds. Just define deepsleep e.g. 1 day = 86400. (wakes up every hour for 0.4 seconds until time reached)

Regarding button presses, there is no chance to capture this during deepsleep. This functionality comes with the ESP32 but is not available with the ESP8266. MQTT messages (Power switch) are partly working. As long as there is only ONE relay and the message is retained after wakeup the status is updated. It seem a limitation of ESP8266 and/or PubSubClient to transmit only the first retained message on a wildcard subscribe. I have checked several libraries but did not get it working at least. A little bit pain-in-the-... , because there where much more ideas possible if you can work on ALL topics that came from retained messages after a wildcard subscribe.
My use-case is "measuring". E.g. moisture, water-level and so on. This works fine.

ionciubotaru, 2017-10-09:
Good improvement.
Tasmota is the best interface for ESP8266. I think is time to port it to ESP32 also

PeggyFree, 2017-10-09:
Support for "ultradeep" and  Ultrasonic distance measurement HC-SR04 are great improvements, maybe you can pull this on Tasmota?

Skippy20, 2017-12-08:
Hi,stefanbode
At first my english is bad ;)

Can you Help my in OpenHab2, to send the deepsleep in a different number.

I read the DeepSlepp out with JSONPATH

{ mqtt="<[broker:tele/wemos1/STATE:state:JSONPATH($.DeepSleep)]"}

But what do I have to do to change this number via openhab?

My problem is that it's in a JSONPATH, I do not know how to change it then.

Thanks for the help

davidelang, 2017-12-08:
you would need to create a command to send to the device, see the commands page.

JSONPATH is just a tool to extract one JSON field out of a JSON message that is 
in a variable, it's a function, not a variable.

stefanbode, 2017-12-09:
Davidlang is correct I assume in your case you have to send for example to cmnd/wemos1/deepsleep the value 3600 to get it updated to 1h deepsleep. You can send any command that you can find here in the wiki .let me know if you need more support. Also using openHAB with TASMOTO.

Skippy20, 2017-12-09:
Thank you all. That was the missing thought. It works now.
I will definitely report again if I get problems

Skippy20, 2017-12-09:
:) 
another short question. what does HEAP mean? what can I do with it?

stefanbode, 2017-12-09:
Heap is the current free memory and anything beyond 5000 is a bit critical and some webpages do not function anymore.

Dome2075, 2018-05-27:
Is there now a command for deepsleep?

Skippy20, 2018-05-27:
yes, download this tasmota at
https://github.com/stefanbode/Sonoff-Tasmota/tree/master/sonoff

darkwaiter, 2018-09-28:
Hi all, maybe somebody can help me - I'm not that experienced so far using Sonoff-Tasmota.
I flashed the "firmware_sonoff.bin" from the folder on the ESP8266 and attached a DHT11 to it; running it as "18 generic" module - so far the sensor and the integration in iobroker/sonoff works fine but no temperature or humidity are transmitted. I only receive the "POWER"=switch and "alive"=true signal. Is there anything that I missed?
(I did an inital test with a .bin from arendst and had no problems - the 2 sensorvalues were sent properly)
In a later step I want to activate the deepsleep feature only send the data every 15min.
Thanks for your help...

meingraham, 2019-05-08:
@arnaldob You need to address this question to @stefanbode's forked repository - https://github.com/stefanbode/Sonoff-Tasmota/issues

As a suggestion, you could try a rule [triggered on mqtt#connected](https://github.com/arendst/Sonoff-Tasmota/wiki/Rules#Rule-trigger) to send a [`STATUS 8`](https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#status) message.

Please, address any further questions to Stefan. If you have questions about Tasmota rules, please refer to the [Tasmota Rules wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/Rules) or to the [Tasmota Support Discord Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are best used for Tasmota software feature requests and bug reporting. Troubleshooting is more effective using an interactive forum.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

stefanbode, 2019-05-12:
Hi @arnaldob , I will take a look on Monday to this. There is already a check that it go not into deepsleep before sensor data published. Anyhow I got a similar report from another user. There seems to be situation, where the firmware think is has no sensor and go into deepsleep without sending data. I assume I have to fine tune the detect method. I use it on several sensors in my garden and do not see an issue, but I must admit I do not care if one value is missing, if the next is reported.

MarcinJan, 2019-06-29:
Hello @stefanbode which board is supported with your feature deepsleep. I'm testing it on pure ESP-01 and it's not working (board do not wake up). I found also this:
https://www.instructables.com/id/Enable-DeepSleep-on-an-ESP8266-01/

so I guess soldering is necessary ? What about sonoff basic did you test? Which board you are using in your garden ?

Thanks for answer

MarcinJan, 2019-06-30:
Hello it's me again I done soldering (was relay hard :-)) as mentioned in the link of my last post and deepSleep finally working for me :-D. I wonder which ESP chip version have deepSleep from the beginning I mean without board soldering...

joba-1, 2019-06-30:
basically all except esp-01 :)
-- 
iphone so i typo

MarcinJan, 2019-07-11:
Hello 

Is it somehow possible to define time between deep sleeps ? I mean wake up time ?

Thanks for answer 

Best Regards

stefanbode, 2019-07-11:
Hi, the main branch does not support deepsleep. You have to go to a fork like mine. Search tasmota stefanbode to get there.

MarcinJan, 2019-07-11:
Hello @stefanbode I'm using your branch but I had problems with analog sensor because it wasn't able to send data because of short wake up time so I was wandering if it's possible to define it? My sensor is working now but I change AdcRead to take  only one sample.

stefanbode, 2019-07-11:
It is all defines by teleperiod. The minimum is 10sek, but you can increase it. Normally there should be no issue because adc read is very fast and do not need extra time.

MarcinJan, 2019-07-12:
@stefanbode thanks for hint. I change teleperiod from 10s to 20s and it helps me as it is shown on the plot (I didn't AdcRead factor which is 5 ) 

[image]

stefanbode, 2019-07-16:
If it happens at 10sek so often you should connect serial and see what the log is telling. I have implemented several checks that only send the device to deelspleep again if there is no connection possible to the network or mqtt server. If all these devices are ok, it is in my view very unlikely it will not publish the data. Please set seriallog 5 to get most info.

meingraham, 2019-07-16:
You may want to set up a syslog server so you can trap all log messages. Especially with higher logging levels, unless you are interactively monitoring the Console, some log messages may scroll off.

stefanbode, 2019-07-16:
Syslog if fine but only wOrks if WiFi is connected. Because with 10sec it is very easy to reproduce in this case my idea was to use seriallog. Here you get all information. Please share at least 3 unsuccessful starts after deepsleep. Just ensure they look all the same.

meingraham, 2019-07-16:
`only wOrks if WiFi is connected` - Duh! :man_facepalming: