arendst/Tasmota #950: Need some help with OTA

arun2118, 2017-10-02:
Been reading https://github.com/arendst/Sonoff-Tasmota/wiki/Arduino-IDE , the section about "Optional: Prepare for OTA upload". I know i'm not a noob and understand this is a noob question but what webserver? Do i need to install Python 2.7 or is the one 3.xx ok? On it i run that .py script then in IDE i can upload?

ionciubotaru, 2017-10-02:
1. upload the software via serial
2. go to the web interface and upload the new version manualy

jetema, 2017-10-02:
OTA is done through web interface, look at the pictures as example
[image]
[image]

arun2118, 2017-10-02:
I see, not through arduino IDE.  But then why add a option for it in IDE settings?

arendst, 2017-10-02:
The option allows to upload your compiled code to a webserver. This webserver then provides the compiled code to your devices if you execute the mqtt upgrade command initiated on your device.

Very handy to upgrade multi devices at once using a single mqtt group upgrade command.

arun2118, 2017-10-02:
How do you get a bin file?

jetema, 2017-10-03:
The bin file is available on   https://github.com/arendst/Sonoff-Tasmota/releases

PeggyFree, 2017-10-04:
I was wondering : is it really possible to make batch (group) updates ? If
I'm not wrong, at first WE would need to batch update the minimal bin.
Then, when successful, batch flash the full bin. The problem being that my
sonoffs never succeed upgrading at first try when doing ota update via the
web interface. Is it not linked with wifi signal, I have not explanation.

Last thing, it is recommended to save and restore configuration file,
during the update process, especially in production environnent.

All this seems hard to automate to me.

lgnd33, 2017-10-04:
I have a similar problem / requirement, so I'm looking to automate it using
Ansible.
Once I get to something that makes sense, I'll commit it to separate
directory.

wasi54, 2017-10-04:
OTA upgrade by URL does not work. I already have my own webservers with the latest Tasmota built with Atom.
Download test firmware from my webserver is OK, see below. Upgrade the same firmware by file upload works.
Do i have to enable upgrade by URL somewhere?

[image]

~$ wget http://ota/sonoff/firmware.bin
--2017-10-04 10:46:30--  http://ota/sonoff/firmware.bin
Auflösen des Hostnamen »ota (ota)«... 172.16.0.95
Verbindungsaufbau zu ota (ota)|172.16.0.95|:80... verbunden.
HTTP-Anforderung gesendet, warte auf Antwort... 200 OK
Länge: 481600 (470K) [application/octet-stream]
In »»firmware.bin«« speichern.

firmware.bin                            100%[===============================================================================>] 470.31K  --.-KB/s    in 0.001s

lgnd33, 2017-10-04:
Can tasmota resolve "ota" host?
Maybe you should try with IP address first, to see if it's really broken.

wasi54, 2017-10-04:
You're right. Tasmota can't resolv ota host. Don't know why. But usinf the IP Adress has solved the problem!
Many thanks to Ignd33

[image]

wasi54, 2017-10-04:
Additional Information: Tasmota is able to resolve the MQTT host successfully but not the OTA host. It is only one nameserver configured and the nameserver knows it and all other systems are knowing the IP of OTA host:
~$ ping ota
PING ws.staffeln (172.16.0.95) 56(84) bytes of data.
64 bytes from Compaq-8100 (172.16.0.95): icmp_seq=1 ttl=64 time=0.031 ms

arendst, 2017-10-04:
On/Off topic regarding sonoff-minimal: I also read somewhere that if you want to upgrade Tasmota you should first load sonoff-minimal.bin and then sonoff.bin. Dependinf on the size of sonoff.bin this is NOT TRUE.

As long as the Program Size of sonoff.bin is smaller than Free Program Space there is NO NEED to load sonoff-minimal first before loding sonoff.bin as there is still enough space available to load sonoff.bin in one go.

[image]

I provide sonoff-minimal for future use when a newer version of esp8266 core for Arduino is released (2.4.0) which uses at least 24k more code space then the current released version 2.3.0.

Regarding OTA batch updates. That works just fine (also using dns name as long as your dns solves it's name)! I always update some 30 devices at once using command ``upgrade <new version>`` like ``upgrade 5.8.0e`` to the group of devices like ``cmnd/sonoffs/upgrade 5.8.0e``. All 30 devices start accessing the local webserver trying to download the firmware. Sometimes some fail to download and just boot up in the previous version so executing the same command will start downloading for these devices again. The once that did upgrade will not download again as I told them only to upgrade if their version is below ``5.8.0e``. See wiki regarding upgrade syntax.

Backing up configurations is a safeguard I never use but if upgrade from version way back I recommend to use it as it allows to fallback to your previous version as the latest version might have upgraded some settings too.

Another tricky part is if you once changed the define CFG_HOLDER and then install a pre-compiled version your settings will be reset as the pre-compiled version holds another value for CFG_HOLDER. Work around is to change CFG_HOLDER once for your desired config changes and then set it back with the same config changes. From then on you can safely use the pre-compiled binaries as they see that CFG_HOLDER is equal and keep using your saved settings.

Again another problem is that binaries can not be downloaded from Github by Tasmota OTA as the ESP8266 core for Arduino software does not support HTTPS for OTA (yet). Therefore @smadds provides binaries you can download via HTTP. See wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Upgrade) for correct urls.

davidelang, 2017-10-05:
On Wed, 4 Oct 2017, wasi54 wrote:

> Additional Information: Tasmota is able to resolve the MQTT host successfully
> but not the OTA host. It is only one nameserver configured and the nameserver
> knows it and all other systems are [...]

it's probably because 'ota' isn't a fully qualified name and your other systems 
are adding a local domain to ota before doing the query, while tasmota isn't 
smart enough to do that (it's an added complication), so it needs you to give it 
the full domain.

closed 2018-05-07 by stale[bot]