arendst/Tasmota #952: Sonoff SC v2.0 upload error

tomasroud, 2017-10-02:
Hi,

I am trying to upload firmware to Sonoff SC.
I hacked tens of POWs and THs and Touches, SC not :(

I removed TX jumper, as on basic or pow pressed button when connect tu USB.
I tried to swap TX and RX pins and swap back.

Everytime I have upload error in Atom PlatformIO.

Sonoff SC is V2.0 written on the main board.

If I start SC without pressing button, it normally connect to wifi and works, so bad volatge is not the problem.

Could anybody help me pls?
[image]

alton76, 2017-10-02:
I had similar issue as yours using Atom platform io, I was able to fix the issue using Arduino IDE.

Hardware check list: meter the header to see if your getting 3.3v on the FTDi USB to Serial board. 

Arduino setup prerequisite

Board: "Generic ESP8266 Module"
Flash Mode: "DOUT"  <<<<!!!!
Flash Frequency: "40MHz"
Upload Using: "Serial"
CPU Frequency: "80MHz"
Flash Size: "1M (no SPIFFS)"  <<<<!!!!
Debug Port: "Disabled"
Debug Level: "None"
Reset Method: "ck"
Upload Speed: "115200"
Port: Your COM port connected to sonoff

More info
https://github.com/arendst/Sonoff-Tasmota/wiki/Arduino-IDE

alton76, 2017-10-02:
Also try swapping the RX, TX pins some USB to serial boards are setup differently. By reversing the pins, wouldn’t destroy the Sonoff however,  giving 5V to a 3.3V device will. Pay close attention to the voltage. 

Eg: RX - TX, TX - RX

OR

TX - TX, RX - RX

tomasroud, 2017-10-02:
alton76, I tried to swap rx-tx, Iam going to try ArduinoIde, but I thought, if other sonoff devices whack working, SC should work too. I will report after try.

davidelang, 2017-10-02:
check that you are using DOUT, not DIO

tomasroud, 2017-10-02:
Arduino IDE doesn't work too :(
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed

Same errors as PlatformIO wrote. 
Device should be in flash mode, because led is normally blinking, but in flashing mode not.

Voltage is 3,3 thx for attention. 
When I disconnect and power up device, it normally works in eWeLink :(

tomasroud, 2017-10-03:
yes DOUT is set

c2342, 2017-10-04:
I had a similar issue with my SC v2.0 - ones I connected my old FTDI to my SC the serial device disappeared on my laptop and flashing failed. I ended up buying a new FTDI to usb controller (have a look at ITEAD's Foca) and it worked w/o issues. Not 100% sure what the issue is, I previously used the same FTDI controller to flash various SONOFFs but I suspect the SC draws too much power for some cheap serial adapters?!

tomasroud, 2017-10-05:
SOLVED ..... I broke my usb programmer. Maybe I swap 3,3 and GND and fired TX pin. So RX was OK and serial communication on flashed devices was OK.

I found secured steps for all devices:
1) first connect 3,3 and GND pins ... ONLY
2) if device lives .... LED is flashing, light up, etc ... you have right 3,3 and GND pins
3) if device doesn't live ... swap 3,3 and GND pins and continue to step 2)
4) if you have right 3,3 and GND, you can connect TX + RX without wsitching device off
5) if upload doesn't work, swap TX and RX and try again without switching device off
6) be fast!!!, remove tx and rx pins as soon as possible, because sonoff devices pulse voltage and current to USB programmers TX and RX and it can heat and destroy it

I hope this helps someone, because I destroyd 5 USB programmers till find the way :)

closed 2017-10-05 by tomasroud

PavelGorbanj, 2018-10-20:
I also had issue with flashing SC - it was solved by connecting TX to TX and RX to RX. 
All the rest sonoffs I've recently flashed had "proper" TX-RX and RX-TX mappings.