arendst/Tasmota #954: Sonoff RF goes to defaults once RF button is pressed for a long time

foxel, 2017-10-02:
Hi.

My issue is as follows:
My child pressed and held down remote button for about 10 seconds and then module became inaccessible.

I've investigated the issue and it looks like the problem is some firmware reset cycle being driven by long press.
```
00:02:33 MQTT: stat/sonoff/RESULT = {"Reset":"Reset and Restarting"}
00:02:33 Cnfg: Use defaults
00:02:40 SYSL: Syslog Host not found so logging disabled for 600 seconds. Consider syslog 0
```

I think for RF module this cycle should be something different since schematics of RF is pretty funny and pressing RF button seems to be equivalent to pressing body button.

arendst, 2017-10-02:
Play with commands SetOption1 0/1 and SetOption13 0/1. See Wiki.

foxel, 2017-10-02:
@arendst thanks!

SetOption1 works like a charm!

closed 2017-10-30 by arendst