arendst/Tasmota #957: espcomm_upload_mem failed

jeylites, 2017-10-03:
Any idea what this is?

Tasmota ver: 5.8.0

```
Archiving built core (caching) in: /var/folders/xv/crvzgz1524sd7__nxwb8ttrw0000gn/T/arduino_cache_689350/core/core_esp8266_esp8266_generic_UploadTool_esptool,CpuFrequency_80,FlashFreq_40,FlashMode_dout,UploadSpeed_115200,FlashSize_1M0,ResetMethod_ck,Debug_Disabled,DebugLevel_None_____dbfd31d0555bcd0761520a5858d44b80.a
Sketch uses 477912 bytes (46%) of program storage space. Maximum is 1023984 bytes.
Global variables use 43924 bytes (53%) of dynamic memory, leaving 37996 bytes for local variables. Maximum is 81920 bytes.
Uploading 482064 bytes from /var/folders/xv/crvzgz1524sd7__nxwb8ttrw0000gn/T/arduino_build_451341/sonoff.ino.bin to flash at 0x00000000
warning: espcomm_send_command: didn't receive command response
An error occurred while uploading the sketch
warning: espcomm_send_command(FLASH_DOWNLOAD_DATA) failed
warning: espcomm_send_command: wrong direction/command: 0x01 0x03, expected 0x01 0x04
error: espcomm_upload_mem failed

```

maartenpaauw, 2017-10-03:
Just started today with playing around with my Sonoff. I followed the Wiki page and getting the same warnings and errors.

jeylites, 2017-10-03:
@maartenpaauw 

Just wondering, what Arduino IDE version are you using? Everything went bad after upgrading to Arduino 1.8.4 (MAC OSX), so I downgraded to 1.8.3 and it worked. 

Another note, I recently upgraded to macOS High Siera, and  I discovered that the Chinese made CH340G / CH341G is not macOS High Siera compatible. The OS crashes at every plug in.

maartenpaauw, 2017-10-03:
I tried to build and upload with Visual Studio Code and Atom with the Platform.io package installed.
I recently upgraded to Mac OS High Siera aswell.

This is the FTDI I ordered from Ali Express.
https://www.aliexpress.com/item/Free-shipping-FT232RL-FT232-FTDI-USB-3-3V-5-5V-to-TTL-Serial-Adapter-Module-Mini/32351819193.html

When I plugged it in it shows up in the USB list:
[image]

alton76, 2017-10-03:
I tested those FTDI types and it worked however the CH34X types, crashes
the OS. I read some where on the net of a similar problem and a fix, I have
yet to look into it.

maartenpaauw, 2017-10-03:
@alton76 I'm looking forward to your results.

maartenpaauw, 2017-10-03:
**Some extra information.**

Video how I try to get the Sonoff in flash mode:
https://youtu.be/8pD-jmpa5aM

The log/output I get when I try to upload the firmware:
https://gist.github.com/maartenpaauw/a8312b8cc4b01af189aad450d6d67938

The things I configured are the WiFi settings (SSID & password) and uncomment the `env_default = sonoff` line.

I own the Sonoff and FTDI chips 2 times and tried it in multiple combinations.

jeylites, 2017-10-03:
@maartenpaauw 

four things...

1) verify the RX & TX ports

Eg:  FTDI RX ---> TX ESP8266
       FTDI TX ---> RX ESP8266

2) Make sure you have these settings correct
[image]

3) Before you start the uploading process, check to see if Arduino IDE see the FTDI.

4) I had a similar error but after downgrading to Arduino IDE 1.8.3 (Mac OSx) it worked. 

My setup: 
Arduino IDE 1.8.3
macOS Sierra 10.12.6
Tasmota 5.8.0

maartenpaauw, 2017-10-03:
@jeylites 

I tried to flash the firmware with Arduino IDE with the settings displayed in the image above.
Still got the same error (the error you posted).

The only thing different:
My OS is High Sierra (10.13)

Did anyone else successfully flash the firmware with High Sierra?

nicolasbisi, 2017-10-04:
Just to confirm that it is indeed a problem with High Sierra. I was uploading fine and after the update, it was not working anymore. I tried on another computer still running Sierra and it worked fine.

alton76, 2017-10-04:
Could agree more, there is an issue with High Sierra and I have re-confirm
it with a test. Sierra seem a whole lot stable...

At this point, either reinstall to Sierra or install two OS X in one
machine. I haven’t done the dual thing but it’s possible if you want to
keep High Sierra

https://www.google.com/amp/www.macworld.co.uk/how-to/mac-software/how-dual-boot-os-x-el-capitan-macos-sierra-on-mac-3659676/%3Famp

maartenpaauw, 2017-10-04:
I switched to Windows and was able to flash the firmware. I think it is a High Sierra issue.

lgnd33, 2017-10-04:
Would someone care to update Wiki (uploading section) to insert a note
about this?
We can remove if once the issue is solved...

pierrerigal, 2017-10-07:
Same problem here with High Sierra.
Back to Sierra on an other mac, it works fine.

JeremyHX, 2017-10-22:
Same issue - did someone find a workaround?

ol-v-er, 2017-10-23:
Just solved the problem.

Install the VCP drivers for Mac from the FTDI website : http://www.ftdichip.com/Drivers/VCP.htm

After install, reboot (it does not work if you do not reboot).

After reboot, plug the FTDI USB/serial converter. Accept the security alert from MacOS.

Restart the flash process. It works!

PS : Found the info here https://forum.arduino.cc/index.php?PHPSESSID=elctcucgnv8b3nv2k72j2nihc7&topic=427825.15

fcrespo82, 2017-12-13:
Sorry guys! I know that reactions are for what I am saying here. But I NEED to thank @ol-v-er !!

This saved me hours of internet research for this error. I thought I lost my sonoff but tried this and OMFG it worked!!

Thank you very much!

65sc02, 2017-12-22:
THis worked for me, too! Oh my God!!! Thank you soooo much!!

maartenpaauw, 2018-01-01:
If Homebrew is installed and `caskroom/cask` is tapped you can install it with `brew cask install ftdi-vcp-driver`.

tIsGoud, 2018-01-07:
Good team effort, thanks!

In addition to @maartenpaauw I had to install the caskroom drivers:
```
brew tap caskroom/drivers
brew cask install ftdi-vcp-driver
```
After that the uploading worked fine again.

mitchwd, 2018-01-18:
Amazing. Thanks so much @ol-v-er and @tIsGoud!

Surprisingly, I was able to get the upload working over esptool.py (before installing the drivers), but I don't know enough about the differences yet to know why!

johnhopkins, 2018-01-22:
Another thanks to @ol-v-er. This solved the issue for me.

coyte, 2018-01-25:
Same here: High Sierra, esptool.py worked, espcomm failure. Spend a day looking for cause. After installing drivers en replugging USB cable, the issue was solved.
Big thank you to all involved !!!!

closed 2018-06-09 by stale[bot]

teejay-87, 2019-07-09:
Me too, thanks to @tIsGoud  and @ol-v-er

ArthurZhou, 2022-10-01:
i think i have found a solution.
(thaks for this chinese [document](https://www.jianshu.com/p/e4dad46d60ea?u_atoken=29b0100e-81b4-47a6-b71f-8d778a49aabb&u_asession=01MLqETeikKXCCyRetz_pMl9cAjaVIPVf8H0FGQtiY0HKGHWkkKr8ygzKzIEBQcnJhX0KNBwm7Lovlpxjd_P_q4JsKWYrT3W_NKPr8w6oU7K-mu193Q0N2EfoZjGpB_qiVg0pn3tpfEcqG8HZmzd6q3mBkFo3NEHBv0PZUm6pbxQU&u_asig=05QZIoXbmOlEEvUjqc1jGrYXieObKG_i7RqznJ5qNAQYOGNabQQ88gK8MPEc74hWGJOKqwRyY2YEinf2WgMluZTrn4bp4YZdkIHC8bzFWMAkRUpKJJ1UAhVJaJi6HCAHRjk_-2myOXKgENZXnq1NnBc8iqLyFuCxRH_Aqv0oi2d0r9JS7q8ZD7Xtz2Ly-b0kmuyAKRFSVJkkdwVUnyHAIJzRvvHwboHfNwXEQBndBkifEMynamqbaW2lIfElSgHNON6xaDswPo-3_59so9Oh9f1-3h9VXwMyh6PgyDIVSG1W8ccXNt0AyKb9_npzB18lhUA230bFxMjmd6Suvt33CXvl4lMuZILuUEArN32BO-FBWuT2eT5sa0D-GiHcA_oEwKmWspDxyAEEo4kbsryBKb9Q&u_aref=V4NHm4%2BZk9Eh0I1tVdOV5qliWSo%3D))
***
1. go to /Users/<username>/Library/Arduino15/packages/esp8266/tools/esptool (path on macos, im not sure where is it on windows)
2. delete folder `0.4.9`

3. then download [esptool-0.4.12](https://objects.githubusercontent.com/github-production-release-asset-2e65be/28329151/4e3959fe-b144-11e7-8c8f-ebb58bc95de6?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20221001%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20221001T080332Z&X-Amz-Expires=300&X-Amz-Signature=6463ddca2c858eb82eb6428e0b01451b47a668505640e652911dcf4ea904b091&X-Amz-SignedHeaders=host&actor_id=89689293&key_id=0&repo_id=28329151&response-content-disposition=attachment%3B%20filename%3Desptool-0.4.12-osx.tar.gz&response-content-type=application%2Foctet-stream) and extract it
4. copy the folder you've just extracted to the folder you've opened before and rename it to `0.4.9`
5. try to flash again