arendst/Tasmota #958: MQTT server does not connect

rites1978, 2017-10-03:
[image]
[image]
[image]
[image]
[image]
I have read and read so many topics but nothing helped.
I am using raspberry pi3 as a bridge. nodejs is running but it just cannot connect to mqtt server. 
I am new to all this and i have followed all the tutorials to flash sonoff with atom platformio. 
https://www.youtube.com/watch?v=3RmuXn8eS9s&feature=youtu.be
and all the related videos, but i am unable to make this work together. 
can anyone help

arendst, 2017-10-04:
Your flashing is fine!
It just cannot find your MQTT server at 192.168.0.69. Are you running a MQTT server (Probably Mosquitto on RPI) on that IP address?
Pls provide output of command ``status 0``

rites1978, 2017-10-04:
I did not install mosquito as nowhere in the tutorial it says i have to do that as well. 
After your response i tried to that and i am getting this error:

** sudo apt-get install mosquitto
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 mosquitto : Depends: libssl1.0.0 (>= 1.0.1) but it is not installable
E: Unable to correct problems, you have held broken packages.

2: sudo apt-get install mosquitto-clients
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 mosquitto-clients : Depends: libmosquitto1 (= 1.4.14-0mosquitto1~nows1) but it is not going to be installed
E: Unable to correct problems, you have held broken package

arendst, 2017-10-04:
Search the net how to install mosquitto on your RPI

rites1978, 2017-10-04:
i followed all the steps online but this is the error i am getting 
http://www.instructables.com/id/Installing-MQTT-BrokerMosquitto-on-Raspberry-Pi/

arendst, 2017-10-04:
See the questions at the end of that page regarding installing Mosquitto on Stretch

mortenmathiasen, 2017-10-15:
I have the same issue on Mosquitto which is installed just fine on RPI. I can without problems connect to the Mosquitto broker from MQTT.FX. The broker is open which means that no username or password should be applied. I suspect the problem with access from Sonoff to be that a default username and password is forced automatically which makes Mosquitto reject the requests from Sonoff.

rites1978, 2017-10-15:
I got it working. 
The steps I followed were, fresh installation of Raspbian. 
Step 1
1. Install fresh Raspbian copy 
2. Sudo apt-get update
3. Sudo apt-get upgrade

4. Install Mqtt server 
(Look online, I forgot where I downloaded from)
Test MQTT 
4. Hap-node.js installation steps 

https://www.youtube.com/watch?v=3RmuXn8eS9s&feature=youtu.be

5. Sonoff home kit using hombridge

https://www.youtube.com/watch?v=M4qHXoAyrRs

6. Adding additional Sonoff 

Link for adding accessory : 
https://github.com/legotheboss/YouTube-files/wiki/(RPi)-Add-additional-Sonoffs-to-HAP-NodeJS-HomeKit

I did lot of research and tried Atleast 10 times. This is the best way to do it. Hope it works as I can understand how irritating it is to read online specially when they are not in order. 
Hope it works

bstory2001, 2018-03-08:
MQT: Connect failed to openhabianpi:1883, rc -2. Retry in 10 sec

I installed Mosquitto on the RasPI3 via the openhasbianpi configuration tool.  I used MQTT.fx on my PC to verify connection and successfully tested publishing and subscribing.  I followed @mortenmathiasen's hunch and set a password for mosquitto on my PI.  Unfortunately it didn't fix the problem.  MQTT.fx connected fine after I entered the proper credentials but the sonoff still will not connect.  I didn't install any of the stuff @rites1978 mentioned.  Maybe someone can see something useful in my status 0 output? 

04:43:48 MQT: Attempting connection...
04:43:53 MQT: Connect failed to openhabianpi:1883, rc -2. Retry in 10 sec
04:43:53 CMD: status 0
04:43:53 RSL: stat/home/1f/kitchen/light/cabinetli/STATUS = {"Status":{"Module":1,"FriendlyName":"SOKitchenCabinetLight","Topic":"/home/1f/kitchen/light/cabinetli","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
04:43:53 RSL: stat/home/1f/kitchen/light/cabinetli/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://domus1:80/api/arduino/sonoff.ino.bin","Uptime":"0T00:02:30","Sleep":0,"BootCount":23,"SaveCount":68,"SaveAddress":"F9000"}}
04:43:53 RSL: stat/home/1f/kitchen/light/cabinetli/STATUS2 = {"StatusFWR":{"Version":"5.12.0","BuildDateTime":"2018-02-23T21:18:29","Boot":6,"Core":"2_4_0","SDK":"2.1.0(deb1901)"}}
04:43:53 RSL: stat/home/1f/kitchen/light/cabinetli/STATUS3 = {"StatusLOG":{"SerialLog":3,"WebLog":2,"SysLog":0,"LogHost":"domus1","LogPort":514,"SSId1":"","SSId2":"indebuurt2","TelePeriod":300,"SetOption":"00000009"}}
04:43:53 RSL: stat/home/1f/kitchen/light/cabinetli/STATUS4 = {"StatusMEM":{"ProgramSize":509,"Free":492,"Heap":21,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
04:43:53 RSL: stat/home/1f/kitchen/light/cabinetli/STATUS5 = {"StatusNET":{"Hostname":"/home/1f/kitchen/light/cabinetl","IPAddress":"192.168.0.6","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"","Mac":"","Webserver":2,"WifiConfig":2}}
04:43:53 RSL: stat/home/1f/kitchen/light/cabinetli/STATUS6 = {"StatusMQT":{"MqttHost":"openhabianpi","MqttPort":1883,"MqttClientMask":"SO-LKit1","MqttClient":"SO-LKit1","MqttUser":"openhabian","MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
04:43:53 RSL: stat/home/1f/kitchen/light/cabinetli/STATUS7 = {"StatusTIM":{"UTC":"Thu Mar 08 03:43:53 2018","Local":"Thu Mar 08 04:43:53 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1}}
04:43:53 RSL: stat/home/1f/kitchen/light/cabinetli/STATUS10 = {"StatusSNS":{"Time":"2018-03-08T04:43:53"}}
04:43:53 RSL: stat/home/1f/kitchen/light/cabinetli/STATUS11 = {"StatusSTS":{"Time":"2018-03-08T04:43:53","Uptime":"0T00:02:30","Vcc":3.403,"POWER":"OFF","Wifi":{"AP":1,"SSId":"","RSSI":76,"APMac":""}}}
04:44:04 MQT: Attempting connection...
04:44:09 MQT: Connect failed to openhabianpi:1883, rc -2. Retry in 10 sec
04:44:20 MQT: Attempting connection...
04:44:25 MQT: Connect failed to openhabianpi:1883, rc -2. Retry in 10 sec
04:44:36 MQT: Attempting connection...

bstory2001, 2018-03-08:
Update.  It connects when I use the ip address of my PI rather than the hostname.  Is it possible to get it working using the hostname?

stepanovalex, 2018-04-20:
I have the same problem :(
Some modules can not connect to the MQTT (sometimes):

11:43:36 WIF: Checking connection...
11:43:36 WIF: Connected
11:43:44 MQT: Attempting connection...
11:43:49 MQT: Connect failed to 10.40.20.30:1883, rc -2. Retry in 10 sec
11:44:00 MQT: Attempting connection...
11:44:05 MQT: Connect failed to 10.40.20.30:1883, rc -2. Retry in 10 sec

Another module at the same time with the same settings is connected and running fine.

Same wifi-network, same mqtt-server, same login/pass for mqtt

status 0 output:
11:37:07 RSL: stat/LSW-bathroom/STATUS = {"Status":{"Module":30,"FriendlyName":"Ванная","Topic":"LSW-bathroom","ButtonTopic":"0","Power":2,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
11:37:07 RSL: stat/LSW-bathroom/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://10.40.20.30:82/api/arduino/LSW-bathroom.bin","Uptime":0,"Sleep":0,"BootCount":108,"SaveCount":1993,"SaveAddress":"FB000"}}
11:37:07 RSL: stat/LSW-bathroom/STATUS2 = {"StatusFWR":{"Version":"5.11.1h","BuildDateTime":"2018-02-18T11:27:41","Boot":6,"Core":"2_4_0_RC2","SDK":"2.1.0(7106d38)"}}
11:37:07 RSL: stat/LSW-bathroom/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"One","LogPort":514,"SSId1":"hn","SSId2":"opushkino.com","TelePeriod":300,"SetOption":"000A0009"}}
11:37:07 RSL: stat/LSW-bathroom/STATUS4 = {"StatusMEM":{"ProgramSize":475,"Free":464,"Heap":22,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
11:37:07 RSL: stat/LSW-bathroom/STATUS5 = {"StatusNET":{"Hostname":"LSW-bathroom-7035","IPAddress":"10.40.20.60","Gateway":"10.40.20.1","Subnetmask":"255.255.255.0","DNSServer":"10.40.20.1","Mac":"DC:4F:22:83:1B:7B","Webserver":2,"WifiConfig":5}}
11:37:07 RSL: stat/LSW-bathroom/STATUS6 = {"StatusMQT":{"MqttHost":"10.40.20.30","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_831B7B","MqttUser":"lsw","MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
11:37:07 RSL: stat/LSW-bathroom/STATUS7 = {"StatusTIM":{"UTC":"Fri Apr 20 08:37:07 2018","Local":"Fri Apr 20 11:37:07 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":3}}
11:37:07 RSL: stat/LSW-bathroom/STATUS10 = {"StatusSNS":{"Time":"2018-04-20T11:37:07"}}
11:37:07 RSL: stat/LSW-bathroom/STATUS11 = {"StatusSTS":{"Time":"2018-04-20T11:37:07","Uptime":0,"POWER1":"OFF","POWER2":"ON","POWER3":"OFF","Wifi":{"AP":1,"SSId":"hn","RSSI":70,"APMac":"64:D1:54:F7:81:AC"}}}

cubilon, 2018-04-20:
I had the same problem. Are you using Homebridge on your network?

stepanovalex, 2018-04-20:
yes, i have domoticz+homebridge+mydomoathome

cubilon, 2018-04-20:
Ok.
I solved finding on my homebridge json configuration file a conceptual mistake:  if you have different switches that control the same tasmota mqtt device, this creates problems like the one you described.

for example, I had something like this:

```
	"activityTopic": "tele/Tavolo/LWT",
	"activityParameter": "Online",
	"startCmd": "cmnd/Tavolo/TelePeriod",
	"startParameter": "120",
```
twice on my config.
removing the duplicity of some MQTT commands/topics on my json config file solved the problem that was driving me crazy.

Hope it helps in some way.

stepanovalex, 2018-04-20:
Unfortunately, not my case.
In my configuration homebridge does not work directly with MQTT - only through Domoticz (via eDomoticz plugin)

pixelos-cc, 2018-06-30:
I stumbled about this topic because I had the problem that one of my two sonoffs would connect to my mqtt server (in this case an iobroker sonoff adapter) while the other one would only give me an error message: 
MQT: Connect failed to someip:1883, rc -2. Retry in 10 sec

Although was sure I had them configured identically it turns out, that in the failing sonoff I had configured the ip-adress of the MQTT server wrong. I pasted the url with the http protocol. 

After changing the server-url from
http://someip:1883
to just:
someip:1883

it worked. 
Maybe this helps someone.

LoveFlashSonoff, 2018-07-22:
@pixelos: Thats solved my problem!
Great stuff.
Thanks a lot.

ascillato2, 2018-07-22:
Hi,

Closing this issue as the lasts comments gave a solution and also there is a confirmation. Please, don't hesistate to ask for reopen this issue if needed. Thanks a lot for everyone that have helped.

closed 2018-07-22 by ascillato2