arendst/Tasmota #959: error: ld returned 1 exit status

alton76, 2017-10-03:
I reformatted my Mac so I had to rebuild my Arduino IDE from scratch. Everything worked before but now I'm getting the following error. Any ideas?

Previously I was using...
Arduino 1.8.3 
Tasmota 5.8.0
MacOs 10.12.6

```
Archiving built core (caching) in: /var/folders/tb/njs756716hx8fc1859g5vn3m0000gn/T/arduino_cache_605419/core/core_esp8266_esp8266_generic_CpuFrequency_80,FlashFreq_40,FlashMode_dio,UploadSpeed_115200,FlashSize_512K64,ResetMethod_ck,Debug_Disabled,DebugLevel_None_____bb93710b0451e34bffb7ff2dbea425ed.a
/Users/Alton/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: sonoff.ino.elf section `.irom0.text' will not fit in region `irom0_0_seg'
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board Generic ESP8266 Module.
```

jeylites, 2017-10-03:
Check the Flash Size, it should be " 1M (NO SPIFFS)"
Ran into a similar issue and it fixed it

alton76, 2017-10-03:
@jeylites it worked, it must have auto default to something else when I reinstalled IDE. Thank you

Jim1948, 2017-10-03:
You have flash mode set to DIO should be DOUT

closed 2018-05-07 by stale[bot]