arendst/Tasmota #961: DHT22 sensor data only published after a visit to the sonoff device web page

joba-1, 2017-10-04:
Hi,

I have a sonoff basic module with current tasmota fw.
Power switching and switch status query with mqtt works fine.

I have now attached a dht22 to gpio14 and configured the pin for dht22 with the devices web page.

If I restart the device, it does not publish sensor data. 
Not in syslog (I configured it to show info):
`switch2 ESP-MQT: stat/switch2/STATUS10 = {"StatusSNS":{"Time":"2017-10-04T03:42:50"}}`
And not in mqtt stat (also only time) and not in mqtt tele (there is no tele/switch2/SENSOR topic published, only tele/switch2/STATE.

If I now go to the devices web page, I immediately see the temp/humidity values there.
And from that moment I also get the values via mqtt stat and tele!

Is that a bug or a configuration error on my side?

I saw a similar (dead?) issue [471](https://github.com/arendst/Sonoff-Tasmota/issues/471). Feel free to follow up there and close here if you prefer.

Regards,
Joachim

joba-1, 2017-10-04:
FWIW: same with DHT11.

arendst, 2017-10-04:
What is the setting of parameter ``TelePeriod`` ? This should be set to some value like 300 which means that at least every 5 muinutes the sensor is initiated and read.

joba-1, 2017-10-04:
I left it on standard (300). Tele is reported every 5 min. Only SENSORS is not.

arendst, 2017-10-04:
Strange. If I execute command ``status 10`` it responds with:
```
14:47:09 MQT: stat/wemos5/STATUS10 = {"StatusSNS":{"Time":"2017-10-04T14:47:09", "SHT1X":{"Temperature":20.8, "Humidity":53.9}, "HTU21":{"Temperature":21.3, "Humidity":53.5}, "BMP280":{"Temperature":22.1, "Pressure":1017.2}, "TempUnit":"C"}}
```
While the normal tele message looks like:
```
14:46:26 MQT: tele/wemos5/STATE = {"Time":"2017-10-04T14:46:26", "Uptime":0, "Vcc":2.670, "POWER":"OFF", "Wifi":{"AP":2, "SSId":"indebuurt2", "RSSI":84, "APMac":"00:22:6B:FE:8E:20"}}
14:46:26 MQT: tele/wemos5/SENSOR = {"Time":"2017-10-04T14:46:26", "SHT1X":{"Temperature":20.7, "Humidity":53.8}, "HTU21":{"Temperature":21.3, "Humidity":53.4}, "BMP280":{"Temperature":22.1, "Pressure":1017.3}, "TempUnit":"C"}
```
Do you see the tele message on the webconsole?

joba-1, 2017-10-04:
your output looks fine to me? Just like I would expect it on mine. But I must open the sonoffs webpage first to get sensor values with status10 or tele. And after that, it works for me, too. I try to restart it and see what is in the weblog

joba-1, 2017-10-04:
Now I struggle with instabilities.

Only every 10th or less try to open a web page succeeds.

I often see LWT offline/online messages.

Status 0 returned heap 8. I guess this is not enough free memory?

Maybe I should go from git to stable version? I‘m ready to test more though...

arendst, 2017-10-04:
heap 8 is way to small and usually caused by use of TLS
Many LWT offline/online mean problematic MQTT server connection
Git or Stable is not the issue. What changes did you make to user_config.h (if any)?

joba-1, 2017-10-04:
no TLS.

git diff |cat
diff --git a/sonoff/user_config.h b/sonoff/user_config.h
index 1df4b42..a83bb2b 100644
--- a/sonoff/user_config.h
+++ b/sonoff/user_config.h
@@ -39,26 +39,26 @@
 
 // -- Wifi ----------------------------------------
 #define WIFI_IP_ADDRESS        "0.0.0.0"         // [IpAddress1] Set to 0.0.0.0 for using DHCP or IP address
-#define WIFI_GATEWAY           "192.168.2.254"   // {IpAddress2] If not using DHCP set Gateway IP address
+#define WIFI_GATEWAY           "192.168.1.222"   // {IpAddress2] If not using DHCP set Gateway IP address
 #define WIFI_SUBNETMASK        "255.255.255.0"   // [IpAddress3] If not using DHCP set Network mask
-#define WIFI_DNS               "192.168.2.27"    // [IpAddress4] If not using DHCP set DNS IP address (might be equal to WIFI_GATEWAY)
+#define WIFI_DNS               "192.168.1.222"    // [IpAddress4] If not using DHCP set DNS IP address (might be equal to WIFI_GATEWAY)
 
-#define STA_SSID1              "indebuurt1"      // [Ssid1] Wifi SSID
-#define STA_PASS1              "VnsqrtnrsddbrN"  // [Password1] Wifi password
-#define STA_SSID2              "indebuurt2"      // [Ssid2] Optional alternate AP Wifi SSID
-#define STA_PASS2              "VnsqrtnrsddbrN"  // [Password2] Optional alternate AP Wifi password
-#define WIFI_CONFIG_TOOL       WIFI_WPSCONFIG    // [WifiConfig] Default tool if wifi fails to connect
+#define STA_SSID1              "job.fritz.ssid"  // [Ssid1] Wifi SSID
+#define STA_PASS1              "helloworld"      // [Password1] Wifi password
+#define STA_SSID2              "JoBa-SE"         // [Ssid2] Optional alternate AP Wifi SSID
+#define STA_PASS2              "helloworld"      // [Password2] Optional alternate AP Wifi password
+#define WIFI_CONFIG_TOOL       WIFI_MANAGER      // [WifiConfig] Default tool if wifi fails to connect
                                                  //   (WIFI_RESTART, WIFI_SMARTCONFIG, WIFI_MANAGER, WIFI_WPSCONFIG, WIFI_RETRY, WIFI_WAIT)
 
 // -- Syslog --------------------------------------
-#define SYS_LOG_HOST           "domus1"          // [LogHost] (Linux) syslog host
+#define SYS_LOG_HOST           "job4"            // [LogHost] (Linux) syslog host
 #define SYS_LOG_PORT           514               // [LogPort] default syslog UDP port
-#define SYS_LOG_LEVEL          LOG_LEVEL_NONE    // [SysLog]
+#define SYS_LOG_LEVEL          LOG_LEVEL_INFO    // [SysLog]
 #define SERIAL_LOG_LEVEL       LOG_LEVEL_INFO    // [SerialLog]
 #define WEB_LOG_LEVEL          LOG_LEVEL_INFO    // [WebLog]
 
 // -- Ota -----------------------------------------
-#define OTA_URL                "http://domus1:80/api/arduino/" PROJECT ".ino.bin"  // [OtaUrl]
+#define OTA_URL                "http://job4:80/api/arduino/" PROJECT ".ino.bin"  // [OtaUrl]
 
 // -- MQTT ----------------------------------------
 #define MQTT_USE               1                 // [SetOption3] Select default MQTT use (0 = Off, 1 = On)
@@ -72,10 +72,10 @@
   #define MQTT_USER            "cloudmqttuser"      // [MqttUser] Mandatory user
   #define MQTT_PASS            "cloudmqttpassword"  // [MqttPassword] Mandatory password
 #else
-  #define MQTT_HOST            "domus1"          // [MqttHost]
+  #define MQTT_HOST            "job4"            // [MqttHost]
   #define MQTT_PORT            1883              // [MqttPort] MQTT port (10123 on CloudMQTT)
-  #define MQTT_USER            "DVES_USER"       // [MqttUser] Optional user
-  #define MQTT_PASS            "DVES_PASS"       // [MqttPassword] Optional password
+  #define MQTT_USER            "tasmota"         // [MqttUser] Optional user
+  #define MQTT_PASS            "sonoff"          // [MqttPassword] Optional password
 #endif
 
 #define MQTT_BUTTON_RETAIN     0                 // [ButtonRetain] Button may send retain flag (0 = off, 1 = on)
@@ -99,7 +99,10 @@
 // %topic% token options (also ButtonTopic and SwitchTopic)
 #define MQTT_TOPIC             PROJECT           // [Topic] (unique) MQTT device topic
 #define MQTT_GRPTOPIC          "sonoffs"         // [GroupTopic] MQTT Group topic
-#define MQTT_CLIENT_ID         "DVES_%06X"       // [MqttClient] Also fall back topic using Chip Id = last 6 characters of MAC address
+#define MQTT_CLIENT_ID         "SOnOff_%06X"  
+
+
+// [MqttClient] Also fall back topic using Chip Id = last 6 characters of MAC address
 
 // -- MQTT - Telemetry ----------------------------
 #define TELE_PERIOD            300               // [TelePeriod] Telemetry (0 = disable, 10 - 3600 seconds)

arendst, 2017-10-04:
Pls provide output of ``status 0``

joba-1, 2017-10-04:
```
cmnd/switch2/status 0
stat/switch2/STATUS {"Status":{"Module":1, "FriendlyName":"Switch 2", "Topic":"switch2", "ButtonTopic":"0", "Power":0, "PowerOnState":3, "LedState":1, "SaveData":1, "SaveState":1, "ButtonRetain":0, "PowerRetain":0}}
tele/switch2/LWT Offline
tele/switch2/LWT Online
cmnd/switch2/POWER (null)

```

not what you have hoped for, I guess. I'll try harder...

arendst, 2017-10-04:
Ha ;-)
I expected something like this:
```
17:35:45 CMD: status 0
17:35:45 MQT: stat/wemos5/STATUS = {"Status":{"Module":18, "FriendlyName":"Wemos5", "Topic":"wemos5", "ButtonTopic":"", "Power":0, "PowerOnState":3, "LedState":1, "SaveData":1, "SaveState":1, "ButtonRetain":0, "PowerRetain":0}}
17:35:45 MQT: stat/wemos5/STATUS1 = {"StatusPRM":{"Baudrate":115200, "GroupTopic":"sonoffs", "OtaUrl":"http://domus1:80/api/arduino/firmware.bin", "Uptime":0, "Sleep":0, "BootCount":32, "SaveCount":77, "SaveAddress":"F7000"}}
17:35:45 MQT: stat/wemos5/STATUS2 = {"StatusFWR":{"Version":"5.8.0f", "BuildDateTime":"2017-10-04T17:06:54", "Boot":31, "Core":"2_3_0", "SDK":"1.5.3(aec24ac9)"}}
17:35:45 MQT: stat/wemos5/STATUS3 = {"StatusLOG":{"SerialLog":3, "WebLog":2, "SysLog":2, "LogHost":"domus1", "LogPort":514, "SSId1":"indebuurt1", "SSId2":"indebuurt2", "TelePeriod":300, "SetOption":"5580280B"}}
17:35:45 MQT: stat/wemos5/STATUS4 = {"StatusMEM":{"ProgramSize":472, "Free":528, "Heap":23, "ProgramFlashSize":1024, "FlashSize":4096, "FlashMode":3}}
17:35:45 MQT: stat/wemos5/STATUS5 = {"StatusNET":{"Hostname":"wemos5", "IPAddress":"192.168.2.200", "Gateway":"192.168.2.254", "Subnetmask":"255.255.255.0", "DNSServer":"192.168.2.27", "Mac":"5C:CF:7F:13:9F:3D", "Webserver":2, "WifiConfig":5}}
17:35:45 MQT: stat/wemos5/STATUS6 = {"StatusMQT":{"MqttHost":"domus1", "MqttPort":1883, "MqttClientMask":"DVES_%06X", "MqttClient":"DVES_139F3D", "MqttUser":"DVES_USER", "MAX_PACKET_SIZE":512, "KEEPALIVE":15}}
17:35:45 MQT: stat/wemos5/STATUS7 = {"StatusTIM":{"UTC":"Wed Oct 04 15:35:45 2017", "Local":"Wed Oct 04 17:35:45 2017", "StartDST":"Sun Mar 26 02:00:00 2017", "EndDST":"Sun Oct 29 03:00:00 2017", "Timezone":99}}
17:35:45 MQT: stat/wemos5/STATUS10 = {"StatusSNS":{"Time":"2017-10-04T17:35:45", "SHT1X":{"Temperature":21.1, "Humidity":53.8}, "HTU21":{"Temperature":21.8, "Humidity":53.4}, "BMP280":{"Temperature":22.6, "Pressure":1015.8}, "TempUnit":"C"}}
17:35:45 MQT: stat/wemos5/STATUS11 = {"StatusSTS":{"Time":"2017-10-04T17:35:45", "Uptime":0, "Vcc":2.670, "Wifi":{"AP":2, "SSId":"indebuurt2", "RSSI":86, "APMac":"00:22:6B:FE:8E:20"}}}
```
You might try the web console and enter ``status 0``

joba-1, 2017-10-04:
Webserver access is rarely working. Good enough?

```
for ((m=1; m<12; m++)); do mosquitto_pub -m $m -t cmnd/switch2/status; sleep 1; done

cmnd/switch2/status 1
stat/switch2/STATUS1 {"StatusPRM":{"Baudrate":115200, "GroupTopic":"sonoffs", "OtaUrl":"http://domus1:80/api/arduino/sonoff.ino.bin", "Uptime":2, "Sleep":0, "BootCount":58, "SaveCount":206, "SaveAddress":"F6000"}}
cmnd/switch2/status 2
stat/switch2/STATUS2 {"StatusFWR":{"Version":"5.8.0c", "BuildDateTime":"2017-09-30T23:02:53", "Boot":6, "Core":"eb891cd6", "SDK":"2.0.0(656edbf)"}}
cmnd/switch2/status 3
stat/switch2/STATUS3 {"StatusLOG":{"SerialLog":2, "WebLog":2, "SysLog":2, "LogHost":"job4", "LogPort":514, "SSId1":"job.fritz.ssid", "SSId2":"JoBa-SE", "TelePeriod":300, "SetOption":"55800009"}}
cmnd/switch2/status 4
stat/switch2/STATUS4 {"StatusMEM":{"ProgramSize":498, "Free":504, "Heap":18, "ProgramFlashSize":1024, "FlashSize":1024, "FlashMode":3}}
cmnd/switch2/status 5
stat/switch2/STATUS5 {"StatusNET":{"Hostname":"switch2", "IPAddress":"192.168.1.167", "Gateway":"192.168.1.222", "Subnetmask":"255.255.255.0", "DNSServer":"192.168.1.222", "Mac":"5C:CF:7F:BD:6A:06", "Webserver":2, "WifiConfig":2}}
cmnd/switch2/status 6
stat/switch2/STATUS6 {"StatusMQT":{"MqttHost":"job4", "MqttPort":1883, "MqttClientMask":"sonoff_%06X", "MqttClient":"sonoff_BD6A06", "MqttUser":"tasmota", "MAX_PACKET_SIZE":512, "KEEPALIVE":15}}
cmnd/switch2/status 7
stat/switch2/STATUS7 {"StatusTIM":{"UTC":"Wed Oct 04 15:41:04 2017", "Local":"Wed Oct 04 16:41:04 2017", "StartDST":"Sun Mar 26 02:00:00 2017", "EndDST":"Sun Oct 29 03:00:00 2017", "Timezone":1}}
cmnd/switch2/status 8
stat/switch2/STATUS {"Status":{"Module":1, "FriendlyName":"Switch 2", "Topic":"switch2", "ButtonTopic":"0", "Power":0, "PowerOnState":3, "LedState":1, "SaveData":1, "SaveState":1, "ButtonRetain":0, "PowerRetain":0}}
cmnd/switch2/status 9
stat/switch2/STATUS {"Status":{"Module":1, "FriendlyName":"Switch 2", "Topic":"switch2", "ButtonTopic":"0", "Power":0, "PowerOnState":3, "LedState":1, "SaveData":1, "SaveState":1, "ButtonRetain":0, "PowerRetain":0}}
cmnd/switch2/status 10
stat/switch2/STATUS10 {"StatusSNS":{"Time":"2017-10-04T16:41:07"}}
cmnd/switch2/status 11
stat/switch2/STATUS11 {"StatusSTS":{"Time":"2017-10-04T16:41:08", "Uptime":2, "Vcc":3.472, "POWER":"OFF", "Wifi":{"AP":1, "SSId":"job.fritz.ssid", "RSSI":76, "APMac":"00:18:39:C6:B6:32"}}}

```

arendst, 2017-10-04:
Looks fine with me.

For debugging purposes you might enable more logging with ``syslog 3`` that might show if the DH22/11 is accessed correctly.

joba-1, 2017-10-04:
btw. yesterday it was working much more stable - I had some issues while integrating this in openhab which was maybe also caused by this, but in general it worked, including status 0 and the web interface. Could it be the device uses more heap over time? Can I somehow bring the device to factory default and start from scratch?

arendst, 2017-10-04:
Reset 1 will reset configuration. Reset 2 will erase flash first then reset to your user_config.h settings. I haven't seen heap grow in the last months; None of my 27 devices have a heap below 22k.

joba-1, 2017-10-04:
I wont reset for now. Want to find out whats going on first, if possible. Syslog 3 gave a hint:
```

2017-10-04T18:06:51.657493+02:00 switch2 ESP-MQT: stat/switch2/RESULT = {"SysLog":"3 (Active 3)"}
2017-10-04T18:06:51.973149+02:00 switch2 ESP-CFG: Saved to flash at F5, Count 207, Bytes 1648
2017-10-04T18:07:33.605989+02:00 switch2 ESP-RSL: Group 0, Index 1, Command STATUS, Data 10
2017-10-04T18:07:33.629255+02:00 switch2 ESP-DHT: Timeout waiting for start signal low pulse
2017-10-04T18:07:33.630504+02:00 switch2 ESP-MQT: stat/switch2/STATUS10 = {"StatusSNS":{"Time":"2017-10-04T17:07:34"}}

```
This timeout is suspicious.

joba-1, 2017-10-04:
some new pieces to the puzzle:

Each time I try to open the web page, the browser hangs forever and heap is ~3kb less.
When down to 8kb free heap. the next web access leads to a reboot with

`"RestartReason":"Fatal exception:3 flag:2 (EXCEPTION) epc1:0x4010011d epc2:0x00000000 epc3:0x00000000 excvaddr:0x4003b870 depc:0x00000000"`

arendst, 2017-10-04:
See my answer before. You are using a not supported version of the esp8266 core for Arduino.

joba-1, 2017-10-04:
Oh, somehow I overlooked that part of your answer: Yes, I use the github version of the core. 
It is not strange, it is documented as one of the possible installation methods for the esp8266 core. But ok, not supported by you, I fully understand that. I use it mainly because of the progress the accompanying libs made regarding "real" https certificate verification. 
I will go back to plain stable 2.3.0 and test if OTA flash works with that low memory. I guess not, so that will take some time...

joba-1, 2017-10-04:
Anyways: Thanks for your great support and software!

joba-1, 2017-10-04:
Good news, somehow: the instability and leak are gone, now that I no longer use an ipsec vpn (avm fritz box). Most probably this has nothing to do with the original issue. 
I‘ll continue looking into that sensor timeout error.
Maybe later I look into why this http server wont work over vpn while everything else does.

pephumin, 2017-10-25:
I also experience the same issue here for two of my Sonoff Basic. May I ask how you solved the problem?

My status 0 is as follows.

```
14:40:28 CMD: Status 0 
14:40:28 MQT: stat/pandora/STATUS = {"Status":{"Module":1, "FriendlyName":"Pandora", "Topic":"pandora", "ButtonTopic":"0", "Power":0, "PowerOnState":3, "LedState":7, "SaveData":1, "SaveState":1, "ButtonRetain":0, "PowerRetain":1}} 
14:40:28 MQT: stat/pandora/STATUS1 = {"StatusPRM":{"Baudrate":115200, "GroupTopic":"pandora", "OtaUrl":"http://192.168.1.100:45873/oh/Pandora.ino.bin", "Uptime":0, "Sleep":0, "BootCount":3, "SaveCount":9, "SaveAddress":"F5000"}} 
14:40:28 MQT: stat/pandora/STATUS2 = {"StatusFWR":{"Version":"5.8.0", "BuildDateTime":"2017-10-25T14:35:13", "Boot":31, "Core":"2_3_0", "SDK":"1.5.3(aec24ac9)"}} 
14:40:28 MQT: stat/pandora/STATUS3 = {"StatusLOG":{"SerialLog":2, "WebLog":2, "SysLog":3, "LogHost":"192.168.1.100", "LogPort":514, "SSId1":"deep", "SSId2":"deeper", "TelePeriod":30, "SetOption":"55800029"}} 
14:40:28 MQT: stat/pandora/STATUS4 = {"StatusMEM":{"ProgramSize":453, "Free":484, "Heap":26, "ProgramFlashSize":1024, "FlashSize":1024, "FlashMode":3}} 
14:40:28 MQT: stat/pandora/STATUS5 = {"StatusNET":{"Hostname":"pandora-1984", "IPAddress":"192.168.1.25", "Gateway":"192.168.1.1", "Subnetmask":"255.255.255.0", "DNSServer":"192.168.1.1", "Mac":"5C:CF:7F:BC:A7:C0", "Webserver":2, "WifiConfig":4}} 
14:40:28 MQT: stat/pandora/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.100", "MqttPort":1883, "MqttClientMask":"pandora", "MqttClient":"pandora", "MqttUser":"hass", "MAX_PACKET_SIZE":512, "KEEPALIVE":15}} 
14:40:28 MQT: stat/pandora/STATUS7 = {"StatusTIM":{"UTC":"Wed Oct 25 07:40:28 2017", "Local":"Wed Oct 25 14:40:28 2017", "StartDST":"Sun Mar 26 02:00:00 2017", "EndDST":"Sun Oct 29 03:00:00 2017", "Timezone":7}} 
14:40:28 MQT: stat/pandora/STATUS10 = {"StatusSNS":{"Time":"2017-10-25T14:40:28", "DHT22":{"Temperature":30.4, "Humidity":47.4}, "TempUnit":"C"}} 
14:40:28 MQT: stat/pandora/STATUS11 = {"StatusSTS":{"Time":"2017-10-25T14:40:28", "Uptime":0, "Vcc":3.145, "POWER":"OFF", "Wifi":{"AP":1, "SSId":"deep", "RSSI":52, "APMac":"68:7F:74:AD:3E:E4"}}} 
14:40:45 MQT: tele/pandora/STATE = {"Time":"2017-10-25T14:40:45", "Uptime":0, "Vcc":3.146, "POWER":"OFF", "Wifi":{"AP":1, "SSId":"deep", "RSSI":52, "APMac":"68:7F:74:AD:3E:E4"}}
```

joba-1, 2017-10-25:
Hi Chesdmethee,

no solution yet.

It took me some time (find out how) to switch to a „supported“ version. It
had the same problem. Then I ran out of time after I discovered that some
http access never returns and keeps a chunk of memory, until there is no
more ram available which causes the sonoff to do all kinds of strange
things and finally reboots.
I hope at some point I find out why the http requests are not answered. It
looks like it is not in tasmota code but in the libraries, but I‘m not sure
about that.

BR,
Joachim

um 09:48:

pephumin, 2017-10-26:
Dear Joachim,

Thanks for your comment with valued information.

While I also still cannot find what cause this issue, I revert back to use the previous version of 5.7.1 which works well (5.8.0 doesn't).

Regards,
Phumin

arendst, 2017-10-30:
Implemented in master release v5.9.0

closed 2017-10-30 by arendst