arendst/Tasmota #962: Tasmota and EWelink (android/iphone app)

Jnpferreira, 2017-10-04:
hi, 
i've been tinkering with my sonoff, since i wanted it to have a physical button i connected it to GPIO14 and installed tasmota (the physical button is 100% working), what i wanted for the sonoff is the app "ewelink" and the physical button. since it has tasmota is it impossible to connect it to ewelink? if not can you explain it to me please?, if so, how can i use something similar? that i can control it via iphone or android.

Thank you

davidelang, 2017-10-05:
I've never heard of ewelink, is it mqtt based? if so, it's trivial to do.

If it's not mqtt based, then tasmota is the wrong software if you want to use 
ewelink. You can have a phone app that talks to openhab if you are just wanting 
a phone app and don't care if it's ewelink.

lgnd33, 2017-10-05:
Ewelink is "original" chinese app that uses proprietary protocol to
communicate with Sonoffs with original firmware.
No use of it with with Sonoff-Tasmota.

ionciubotaru, 2017-10-05:
Ewelink communicate directly with ITEAD's servers. There is no way to connect Tasmota to those servers because is an encripted connection and you don't know the password

Jnpferreira, 2017-10-05:
hi,
thank you for the answers.
Do you know a link/tutorial for what i need to buy/do to use an app in android/iphone? i'm a zero at this, and found some references but can't understand them clearly about what i need to do. all i want is something on the line of ewelink,  ON - OFF, Timer. 

thank you

jetema, 2017-10-05:
Hi Jnpferreira, ewelink app is a "semiautomatic" way to use sonoff,  Tasmota is another diferent and more powerful way to use and control your sonoff device/s in deep.
It depends if you want to use ewelink or Tasmota you have two diferents "paths" to follow and learn.

Jnpferreira, 2017-10-05:
i went to tasmota since i wanted to have a physical button, that part i already made easily (not difficult to begin with), since it's impossible now to work with ewelink, do you know how i can "make/get" an app that works with android and ios, that i can turn on/off, set "alarms" to activate or deactivate lights. there is just too much information/offer here for someone that started this a few days ago. i see ppl talking about mqtt and all, but i don't even know where to start hardware and softwarewise

davidelang, 2017-10-05:
google for setting up openhab (and mosquitto as a mqtt server) on a raspberry pi 
and follow those directions. Then you can use one of the many openhab compatible 
apps on your phone to control things.

Jnpferreira, 2017-10-07:
hi,
done what you've advised, bought a raspberry pi, even managed to prepare it with mosquitto server, and openhab, at least it seems it is 100% ready as i tested with the hello world, and i can access the openhab through my pc.  problem is i can't find a tutorial or something that shows me how i connect/configure the sonoff with the rasp and finally connect it to openhab, all i can find is "tutorials" where the mqtt is already created and configured. is there a tutorial that shows the middle part till the "creation to openhab"? it would be easier for newbies to learn this if there was an A to Z kinda guide/movie. not complaining much lol but information is kinda dispersed or vague to people trying to learn.

sry for the rambling but i've been all day trying to get some decent information for the next step and really can't find it, really stuck on the connection of the sonoff to rasp/mqtt/openhab.

davidelang, 2017-10-07:
connect to the webserver on the sonoff, go to the config section and set the 
server to the ip or dns name of the pi, and the mqtt port to what mosquitto is 
using (should be the default)

harkoy, 2018-03-24:
Hi!
You can use Yeti to manage your sonoff tasmota, they have a beta solution for that.
https://play.google.com/store/apps/details?id=com.netbeast.yeti
 (check that is version 1.4.87 or higher) 

You must have trasmota version 5.10 or higher and no password in the tasmota web console.

I'm still trying to find a way to connect yeti with Google home assistant, thought

demian85, 2018-04-07:
Hey, I've just downloaded and configured Yeti on Android. However, I don't fully understand how the app communicates with the device, as there is no MQTT server configured. Does the app work only when connected to the same WiFi network as the device?

Stevehans, 2018-04-07:
You might find it easier to use something like cloudmqtt and mqtt dashboard, that's similar to ewelink as a concept.

Then use yeti for local lan control.

closed 2018-06-26 by stale[bot]

jdknox, 2020-05-13:
I haven't tested it, but someone has made what appears to be similar to what the OP was looking for: https://github.com/dlashua/ewelink-mqtt

> This project will connect to the eWeLink (iTead/Sonoff) cloud and report device state to MQTT. Additionally, it allows updates received via MQTT to be sent back to the device.

Apparently it isn't too difficult to communicate with iTead's servers in this manner.