arendst/Tasmota #964: Sonoff POW fatal exception.

leodesigner, 2017-10-04:
Hi,
I am getting constant reboots when SWITCH is ON only.

18:03:54 MQT: tele/sonoffPOW/INFO3 = {"RestartReason":"Fatal exception:0 flag:2 (EXCEPTION) epc1:0x40107311 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000"}

Program Version | 5.8.0
-- | --
2017-09-18T16:44:29
2_3_0/1.5.3(aec24ac9)
0 Hours
95 at FB000
17
Exception
Sonoff

Just after few seconds after start.

arendst, 2017-10-04:
Looks like your pows power supply is unable to provide enough power for the logic AND activated relay.

leodesigner, 2017-10-04:
For now the RELAY is ON but the LOAD(LIGHT BULB) is disconnected from the Sonoff POW.
No reboots so far...
Maybe I should close web interface - it might be the reason for reboots (the data streamed to the browser is constantly changing).

leodesigner, 2017-10-04:
Really strange, as soon as I am physically connecting 100W bulb I am getting "Fatal exception".
If no load connected and relay is staying on or off - everything looks stable.
(except Voltage is always 0)

Should I try different version/release of the firmware ?

leodesigner, 2017-10-04:
Flashed 5.7.1
Now it's is stable. So something is changed in the last 5.8.0 release and causing exception.
The Voltage is still 0. Is it possible that Sonoff changed something in hardware ? My two Sonoff POW are dated 2017/5/1

leodesigner, 2017-10-05:
Something is definitely wrong with Sonoff POW power measurement.
I am connected 100W, 1kW and 2kW loads for test.
I was able to see correct values only for 100W bulb. 
For 1kW load I had jumping measurements about 1kW but the value of the current was 0 Amp.
For 2kW load power consumption was jumping between 50W-90W and current was 0 Amp
So basically Power Measurements/Current/Voltage feature is broken for now :(

arendst, 2017-10-05:
Pls provide output of command status 0

leodesigner, 2017-10-05:
```
16:35:34 MQT: /ev/sonoffPOW/stat/STATUS = {"Status":{"Module":6, "FriendlyName":"Sonoff", "Topic":"sonoffPOW", "ButtonTopic":"0", "Power":0, "PowerOnState":3, "LedState":1, "SaveData":1, "SaveState":1, "ButtonRetain":0, "PowerRetain":0}}
16:35:34 MQT: /ev/sonoffPOW/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200, "GroupTopic":"sonoffs", "OtaUrl":"http://domus1:80/api/arduino/sonoff.ino.bin", "Uptime":1, "Sleep":0, "BootCount":33, "SaveCount":171, "SaveAddress":"F7000"}}
16:35:34 MQT: /ev/sonoffPOW/stat/STATUS2 = {"StatusFWR":{"Version":"5.7.1", "BuildDateTime":"2017-09-09T17:28:27", "Boot":31, "Core":"2_3_0", "SDK":"1.5.3(aec24ac9)"}}
16:35:34 MQT: /ev/sonoffPOW/stat/STATUS3 = {"StatusLOG":{"SerialLog":0, "WebLog":2, "SysLog":0, "LogHost":"10.0.1.7", "SSId1":"Open”, "SSId2":"Wrt", "TelePeriod":60, "SetOption":"55800009"}}
16:35:34 MQT: /ev/sonoffPOW/stat/STATUS4 = {"StatusMEM":{"ProgramSize":477, "Free":524, "Heap":25, "ProgramFlashSize":1024, "FlashSize":4096, "FlashMode":3}}
16:35:34 MQT: /ev/sonoffPOW/stat/STATUS5 = {"StatusNET":{"Hostname":"sonoffPOW-6834", "IPAddress":"10.0.11.20", "Gateway":"10.0.1.1", "Subnetmask":"255.255.0.0", "DNSServer":"10.0.1.1", "Mac”:”xx:xx:xx:xx:xx:xx”, "Webserver":2, "WifiConfig":2}}
16:35:34 MQT: /ev/sonoffPOW/stat/STATUS6 = {"StatusMQT":{"MqttHost":"10.0.1.3", "MqttPort":1883, "MqttClientMask":"DVES_%06X", "MqttClient":"DVES_449AB2", "MqttUser":"DVES_USER", "MAX_PACKET_SIZE":512, "KEEPALIVE":15}}
16:35:34 MQT: /ev/sonoffPOW/stat/STATUS7 = {"StatusTIM":{"UTC":"Thu Oct 05 15:35:34 2017", "Local":"Thu Oct 05 16:35:34 2017", "StartDST":"Sun Mar 26 02:00:00 2017", "EndDST":"Sun Oct 29 03:00:00 2017", "Timezone":1}}
16:35:34 MQT: /ev/sonoffPOW/stat/STATUS8 = {"StatusPWR":{"Total":0.135, "Yesterday":0.027, "Today":0.109, "Power":0, "Factor":0.00, "Voltage":0, "Current":0.000}}
16:35:34 MQT: /ev/sonoffPOW/stat/STATUS9 = {"StatusPTH":{"PowerLow":0, "PowerHigh":0, "VoltageLow":0, "VoltageHigh":0, "CurrentLow":0, "CurrentHigh":0}}
16:35:34 MQT: /ev/sonoffPOW/stat/STATUS10 = {"StatusSNS":{"Time":"2017-10-05T16:35:34"}}
16:35:34 MQT: /ev/sonoffPOW/stat/STATUS11 = {"StatusSTS":{"Time":"2017-10-05T16:35:34", "Uptime":1, "Vcc":3.196, "POWER":"OFF", "Wifi":{"AP":1, "SSId":"Open", "RSSI":50, "APMac”:”xx.xx.xx.xx.xx.xx”}}}
```

arendst, 2017-10-05:
Looks ok with me.

leodesigner, 2017-10-05:
Voltage 0 - always, If Power consumption is over 1kW - shows Current 0 A :( and Power consumption report is jumping around 10W-50W - complete noise :(

arendst, 2017-10-05:
As you already tried different versions of tasmota I think it is safe to say you have a hardware problem unless others have same kind of problem...

Mine are still running fine.

leodesigner, 2017-10-05:
Is it possible to return to the original firmware or try other firmwares like Espurna by flashing them via WEB interface ? I want to narrow down the issue. I am still not convinced regarding the hardware problem... Or might they changed something in hardware design. Really strange :(

arendst, 2017-10-05:
You can install any software via serial upload. Via web interface most will fail as they probably use a different memory model.

leodesigner, 2017-10-05:
I wanted to avoid opening the Sonoff case and soldering for now. But I guess I need to go deeper :)
Anyway thanks for help.

PeggyFree, 2017-10-05:
Please note that soldering is not necessary for flashing. Flashing with
dupont cables in the 4 holes is easy.

leodesigner, 2017-10-06:
So, I was able to flash Espurna firmware from Web OTA of Tasmota. It was flawlessly.
However I got the same issue with Voltage, Power and Current.
After flashing second new module everything looks good for now.
So this is a faulty, buggy hardware in this case. Sorry for bothering, thanks for help and great work.

leodesigner, 2017-10-06:
Solved,
Hardware problem.

closed 2017-10-06 by leodesigner