arendst/Tasmota #966: no MQTT services found?

chris-008, 2017-10-04:
good evening,
Being new in this scene, I have been struggling the last past nights with the final step on how to get the sonoff module to work  on an orange pi zero with MQTT broker.

Unfortunately no MQTT services found...
Typical output on the Sonoff console:
[image]

MQTT broker is working though:
[image]

Accessories are installed, working on homebridge   and properly configured:

[image]

[image]
 
the swich is also recognized on the iOS Home app.
but switching doesn't fysically work due to the above MQTT problems.

Anyone who can help me with this one?
Thanks in advance!

chris-008, 2017-10-04:
Portmapping on my BBox3 router is also done:
[image]

chris-008, 2017-10-04:
could it be that the MQTT part in the user_config.h file has to be modified and reflashed? cause that part has been left untouched so far...
[image]

davidelang, 2017-10-05:
connect to the device and see what's configured.

Enable more detailed logging and see what it reports that it's doing.

The problem is probably that the config is not what you think it is (see the 
troubleshooting page)

chris-008, 2017-10-05:
Hello, thanks for your quick reply.
I went through the troubleshooting but did'nt find an issue which is directly related to mine.

Could you inform me on how to get the detailled logging?
Perhaps the  "troubleshooting tools->logs/ syslogs" part?

Should i leave the MQTT par tin the conf file untouched or does it have to be modified accordingly?
(in the different tutorials I found on youtube, it was not mentioned to modify this section though)

jeylites, 2017-10-05:
@chris-008 

Check your MQTT Host on Tasmota, it needs to be the same URL as the broker. 
example:
[image]

@davidelang 

Little off topic but thought will ask here anyway. 

Is it possible to give a nonstatic IP for  MQTT Host on TASMOTA? Can it still find the broker that way?

The reason I'm asking is I'm unable to give a static IP to my Raspberry Pi running MQTT Broker / homebridge because it fails to reconnect after the internet drops out, with DHCP somehow it connects automatically. Very strange and it seems to be a common problem with these PI. There is a few fix out there but I have yet to look into it.

chris-008, 2017-10-05:
Thanks for our help.
MQTT host IP is same as the Pi's:
[image]

I use a username and PW which is also filled in.

davidelang, 2017-10-05:
On Thu, 5 Oct 2017, chris-008 wrote:

> I went through the troubleshooting but did'nt find an issue which is directly related to mine.

see the section 'config problems'

chris-008, 2017-10-05:
Going through the whole Wiki from scratch now...
while trying some more option, i enabled a web password with a known password. When i try to go the sonoff module's IP, I get a verification requirement. Which username should I use here? And is there a way to delete this?

chris-008, 2017-10-05:
reflashing didn't help...

chris-008, 2017-10-05:
ok...note to myself...go through the wiki first, ask later ;-)
After weeks of trying I think I wanted a short cut as I was getting clos to giving up.
but appreciated that you showed me the way!
 
For anyone who ran into this issue:
user_config_h has to be filled in including MQTT parameters.
I made the mistake following a youtube tutorial where this part wasn't mentioned.

Regarding the web password issue: holding ground pin for 4 secs did the trick.

Thanks again for your help guys!

closed 2017-10-05 by chris-008