arendst/Tasmota #967: pwm works but no with dimmer by (setoption15 on)

hamwong, 2017-10-04:
I got no issue with direct pwm when setOpt15 off, but when it's on, led no longer light,
and I found power is keep off, is there anything I am missing?
```
07:16:52 CMD: dimmer 50
07:16:52 MQT: stat/sonoff/RESULT = {"POWER":"OFF", "Dimmer":50}
07:17:58 CMD: dimmer 10
07:17:58 MQT: stat/sonoff/RESULT = {"POWER":"OFF", "Dimmer":10}
07:22:10 CMD: dimmer 20
07:22:10 MQT: stat/sonoff/RESULT = {"POWER":"OFF", "Dimmer":20}
07:22:18 CMD: power on
07:22:18 MQT: stat/sonoff/RESULT = {"Command":"Unknown"}
07:22:30 CMD: power on
07:22:30 MQT: stat/sonoff/RESULT = {"Command":"Unknown"}
```

hamwong, 2017-10-04:
another quesiton, I don't see slider on main menu, does it not allow yet?

hamwong, 2017-10-04:
```
00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_0ED568, GroupTopic sonoffs) Version 5.8.0f
00:00:00 WIF: Connecting to AP1 Ham2 in mode 11N as sonoff-5480...
00:00:04 WIF: Connected
00:00:04 DNS: Initialized
00:00:04 HTP: Web server active on sonoff-5480.local with IP address 192.168.101.112
00:00:06 MQT: Attempting connection...
07:35:45 DNS: Query done. MQTT services found 0
07:35:45 MQT: Connected
07:35:45 MQT: tele/sonoff/LWT = Online (retained)
07:35:45 MQT: cmnd/sonoff/POWER = 
07:35:45 MQT: tele/sonoff/INFO1 = {"Module":"WeMos D1 mini", "Version":"5.8.0f", "FallbackTopic":"DVES_0ED568", "GroupTopic":"sonoffs"}
07:35:45 MQT: tele/sonoff/INFO2 = {"WebServerMode":"Admin", "Hostname":"sonoff-5480", "IPAddress":"192.168.101.112"}
07:35:45 MQT: tele/sonoff/INFO3 = {"RestartReason":"Software/System restart"}
07:35:47 CMD: dimmer
07:35:47 MQT: stat/sonoff/RESULT = {"Dimmer":100}
07:35:51 CMD: dimmer 40
07:35:51 MQT: stat/sonoff/RESULT = {"POWER":"OFF", "Dimmer":40}
07:35:53 MQT: tele/sonoff/STATE = {"Time":"2017-10-05T07:35:53", "Uptime":0, "Wifi":{"AP":1, 
07:35:56 CMD: dimmer 100
07:35:56 MQT: stat/sonoff/RESULT = {"POWER":"OFF", "Dimmer":100}
```

first time system shows only dimmer , than with power off

arendst, 2017-10-05:
Thnx for testing. Errors will be fixed in next (pre)release.

1043717432, 2017-10-05:
hi, what module do you use pwm led lights?

hamwong, 2017-10-05:
i didn't use module, I use Power MOSFET IRLZ44N TO-220 and modify few lights, it make the led light with dimmer.

like below way
[image]

hamwong, 2017-10-06:
v5.8.0g - Fixd, it's works like a charm

closed 2017-10-06 by hamwong

1043717432, 2017-10-08:
Hi, any LED and MOSFET IRLZ44N, can you make the LED darker?

hamwong, 2017-10-08:
yes, it should works, but it depends on your LED power supply, some of the LED drive can't support dimer, when you lower the light, LED start blinking, I didn't modify led strip as it require high voltage