arendst/Tasmota #968: Daylight savings timezone offset for New Zealand is UTC+1300

CliveBull, 2017-10-05:
APP_TIMEZONE seems to be limited to a number in the range -12 to +12 hours. However, New Zealand (and a bunch of Pacific Islands north of us) is currently in daylight saving time, which is UTC+1300. 

I want to use APP_TIMEZONE = 99 to allow for DST but similarly, TIME_DST seems to be limited to an offset of +720 minutes (ie 12 hours x 60 minutes/hour). In line with the above comment, TIME_DST needs to accept a maximum offset of +780 minutes.

arendst, 2017-10-05:
Using Timezone 99 and TIME_DST offsett +780 works fine with me. Did you change define CFG_HOLDER too when you changed the offset as otherwise the changes would not take effect.

I will change Timezone range to -13 and +13 in next (pre)release.

CliveBull, 2017-10-05:
Updated again to check but still having the same issue with TIME_DST. Sonoff is running an hour slow (ie it's on standard time, not daylight savings time). DST/STD rules for NZ are correct:
#define TIME_DST               Last, Sun, Sep, 2, +780
#define TIME_STD               First, Sun, Apr, 3, +720

I'm away now until Sunday so will get back to this then.

arendst, 2017-10-06:
Ah I think what goes wrong. Your winter is my summer (Netherlands) ;-)

Workaround I guess is that you swap TIME_DST and TIME_STD values.

CliveBull, 2017-10-08:
I swapped TIME_DST and TIME_STD and the local time is now correct for NZDT. So just one little change and Tasmota will work for the other 50% of the planet ;)
Thanks Theo - I'm very much enjoying the software.

closed 2017-10-09 by CliveBull

jeylites, 2018-05-11:
How do you go about configuring for countries with No daylight saving?

arendst, 2018-05-11:
@jeylites select a fixed timezone between -13 and 13

jeylites, 2018-05-12:
@arendst 

Just making sure I have it in the right formate. My local offset time is UTC +8 hours all of the period.

`// -- Time - Start Daylight Saving Time and timezone offset from UTC in minutes
#define TIME_DST  +8  // Northern Hemisphere, Last sunday in march at 02:00 +120 minutes

// -- Time - Start Standard Time and timezone offset from UTC in minutes
#define TIME_STD  +8  // Northern Hemisphere, Last sunday in october 02:00 +60 minutes`

arendst, 2018-05-12:
@jeylites Nope.

If you do not observe daylight savings stay off of changing these parameters and just select timezone 8

steven-ellis, 2020-05-22:
You can set the timezone for new zealand on the console via
`
TimeSTD 1,1,4,1,2,720
TimeDST 1,0,9,1,2,780 
timezone 99
`