arendst/Tasmota #969: Syntax of color command

andreasloe, 2017-10-05:
I do not understand the details of the Sonoff B1 color arguments. The wiki says there might be several ways to define the color but I only get a white color. Is there any rational behind the 
 http://192.168.2.13/cm?cmnd=Color%200000009B00
What numbers are necessary? Which numbers define the color?

arendst, 2017-10-05:
The wiki says only one supported Color option for the Sonoff B1:
```
Color | <RRGGBBCCWW> | (Sonoff B1) Set color to RRGGBBCCWW hexadecimal value
```
So a simple command like ``Color 9B00000000`` will turn on the red led. ``Color 009B000000`` should turn on the Green one. You may guess the other options...
All numbers a necessary as they are location specific.

andreasloe, 2017-10-05:
Thanks, understood now. Still, I am wondering how the colors look like. They are not very bright and (for example) it is very difficult to get yellow or the like. If I am entring a particular color the Sonoff replies with a (sometimes slightly) different color. So adding
 http://192.168.2.13/cm?cmnd=Color%20E96F000000&Dimmer=%20100
gives
 RESULT = {"POWER":"on", "Dimmer":91, "Color":"E76E000000"}

closed 2017-10-05 by andreasloe