arendst/Tasmota #970: Configure the MQTT fingerprint from the web configuration

liorsu, 2017-10-05:
I am running tasmota with the Web and encrypted MQTT options (removed the assert, and it is still working fine).
I am using letsencrypt certificates, but it is valid only for three months every time.
So I need to recompile and flash my ESP8266 with a new version of tasmota with the updated fingerprint whenever I have a new certificate.
I would be really helpful to me if it will be possible to change to certificate fingerprint from the web interface. Then I will also be able to create an automation for that and won't need to flash my esp again for that reason.
Thanks a lot Theo for all the great work.
Lior.

arendst, 2017-10-05:
Try to use the web console and command ``MqttFingerprint`` which should do what you want.

liorsu, 2017-10-05:
That is awesome Theo, didn't know that it already exist, thank you so much!
Will it be possible to also include it in the MQTT configuration page of the Web interface?
That way all of the MQTT setup options will be shown there easily, and I think it will even increase the awesomeness level of it!
If you can also look at Issue #808, as me and someone else believe that there might be a potential bug in there, I will also appreciate it a lot!
Thank you so much again for your work and effort invested, I truly appreciate it a lot.
Lior.

stefanbode, 2017-10-07:
@liorsu this is also already implemented :-). You just send to the /cmd/sonoff/MqttFingerprint an EMPTY payload "" and you will get the current Fingerprint. If you send a value, this will change the fingerprint. You are one of the rare users that use the fingerprint like I do. Please add client.stop() in the sonoff.ini near line 636ff.

Should look after modification like this:

```
#ifdef USE_MQTT_TLS
    addLog_P(LOG_LEVEL_INFO, S_LOG_MQTT, PSTR(D_FINGERPRINT));
    if (!espClient.connect(sysCfg.mqtt_host, sysCfg.mqtt_port)) {
      snprintf_P(log_data, sizeof(log_data), PSTR(D_LOG_MQTT D_TLS_CONNECT_FAILED_TO " %s:%d. " D_RETRY_IN " %d " D_UNIT_SECOND),
        sysCfg.mqtt_host, sysCfg.mqtt_port, mqttcounter);
      addLog(LOG_LEVEL_DEBUG);
      return;
    }
    if (espClient.verify(sysCfg.mqtt_fingerprint, sysCfg.mqtt_host)) {
      addLog_P(LOG_LEVEL_INFO, S_LOG_MQTT, PSTR(D_VERIFIED));
    } else {
      addLog_P(LOG_LEVEL_DEBUG, S_LOG_MQTT, PSTR(D_INSECURE));
    }
    espClient.stop();
    yield();
```

liorsu, 2017-10-09:
Thank you @stefanbode, found out about it from your previous issue.
Also interested in the other applications and implementations that you made with it :)
Great minds think alike..

fragolinux, 2017-11-01:
hi, i'm using v5.9.0, ota updated from http://sonoff.maddox.co.uk/tasmota/sonoff.bin
a few days ago i updated al my services to ssl, via letsencrypt, so i used the same certificate for apache, node-red and mosquitto, this last on port 8883... my basic sonoff refuses connection, and via console the mqttfingerprint is just unknown:
18:13:23 CMD: MqttFingerprint
18:13:23 RSL: stat/LampadaStudio/RESULT = {"Command":"Unknown"}

what to do? Thanks :)

arendst, 2017-11-02:
In the provided binaries TLS is disabled as it eats up too much code. Therefore the command Mqtt Fingerprint is disabled too. 

To use TLS you will have to enable it in user_config.h (and disable webserver to make it work within the tight RAM space) and recompile the code.

fragolinux, 2017-11-02:
the 2 functionalities cannot coexist on every board, or just the 1mb sonoffs? if i change the memory on sonoff with a 4mb model, can i have both or have to disable 1 of them anyway? thanks

arendst, 2017-11-02:
You still have to disable one of them as the amount of RAM does not change when replacing the flash.

stefanbode, 2017-11-02:
Yes and no. I run actually the configuration with the webserver and the TLS MQTT. It also run on the 512kb small sonoff devices. I just have to disable some stuff during compilation to make the code fit into the flash. There is ONE thing that you have to keep in mind. It the MQTT TLS is "ON" AND "CONNECTED", you cannot go to the webserver interface and reconfigure the MODULE. This needs to much RAM and crashes. But it is ok to temporary disable MQTT and restart. Then you can make through webserver your MODULE changes. If everything is ok, you just enable MQTT again. Additionally, also for security reasons, I disable in normal operation my webserver. This free up some RAM. If needed I can enable it again with "webserver 2" command through MQTT.

closed 2018-05-07 by stale[bot]