arendst/Tasmota #972: B1 flashed with succes failed connection

sth313, 2017-10-05:
I flashed a Sonoff B1 succesfully with Tasmota but when it tries to connect it doesn't get an IP. It says: 

17:41:50 MQT: Connect failed to domus1:1883, rc -2. Retry in 10 sec
17:42:01 MQT: Attempting connection...
Skipping compressed pointer

I had this before with my S20 and Basic modules but with these device is could push the button a several times till it resets and after that it works.

How to do this with the B1?

closed 2017-10-05 by sth313

nil-ghosh, 2018-01-02:
Hi, I am having the same error. Connect failed to domus1:1883, rc -2. Retry in 10 sec
How did you fix the issue?

sth313, 2018-01-03:
You've reconnect the sonoff and boot it up normal, this should do the trick.