arendst/Tasmota #974: BME280 Calculation of air pressure at sea level

chkbln, 2017-10-05:
Hello Theo,
in the air pressure measurement (BME280 ...) the actual air pressure is measured at the location. The output value for the air pressure corresponds to that for the
current altitude and current temperature. However, the general air pressure data always refers to the sea level. The value is therefore higher for the sites above sea level. Can you install the possibility to enter the altitude at which the sensor is located and thus calculate and output the air pressure at the sea level.
Best regards

arendst, 2017-10-06:
Got it working but uses way to much code due to Arithmetic function pow():
```
double bmp_atSealevel(double pressure)
{
  return pressure / pow(1.0 - ((float)sysCfg.altitude / 44330.0), 5.255);
}
```
+8k !!!!!

Need to think about another solution.....

chkbln, 2017-10-06:
would it be possible to make the function possible via the user_config.h?  So I2C code for BMP / BME280 sensor with altitude or I2C code for BMP / BME280 sensor without altitude

chkbln, 2017-10-08:
Hello Theo
I have compiled the version with the Add Sea level pressure calculation and flashed on the ESP8266. Now the pressure is only output with 0.0 hPa. Is still an error in the adjustment. Where is the possibility to specify the appropriate altitude? Per Console?
Best regards

arendst, 2017-10-09:
Try the "undocumented" command ``altitude <meters>`` ;-)

chkbln, 2017-10-09:
Hello Theo,
with the input of altitude <meters>, the "BMP180 SeaPressure 0.0 hPa" is displayed on the start page. It is working.
Unfortunately, both values are at 0.0. Therefore, no correct value is displayed. The air pressure is currently on Sealevel at approx. 1015 and absolutely at approx. 1007.
In my view, there is still a mistake
Current example with the modified software:
BMP180 Temperature 24.2 ° C
BMP180 Pressure 0.0 hPa
BMP180 SeaPressure 0.0 hPa

Best regards

arendst, 2017-10-09:
I think I know the problem. Will fix tomorrow.

arendst, 2017-10-09:
You might want  to try adding a break; after line 406 in file xsns_bmp.ino like on line 387.

arendst, 2017-10-10:
Workaround works fine. Will implement in next (pre)release.

chkbln, 2017-10-10:
Hello Theo
I have tested it and it works great.
Many Thanks

arendst, 2017-10-30:
Implemented in master release v5.9.0

closed 2017-10-30 by arendst