arendst/Tasmota #975: missing a unique identifier in JSON messages send by the same device

jhlweb, 2017-10-05:
Hi I've loading this project onto multiple wemos board around the house. 
They all send their data to MQTT but there is no way of knowing which MQTT messages are from which sensor.

This is a message from a Wemos D1 mini with 1 DHT22 and 3 DS18B20 sensors.
`tele/sonoff/SENSOR = {"Time":"2017-10-05T22:39:45", "DS18x20":{"DS1":{"Type":"DS18B20", "Address":"28FF4CBFA41604C4", "Temperature":25.37}, "DS2":{"Type":"DS18B20", "Address":"28FF1E7FA116035D", "Temperature":30.44}, "DS3":{"Type":"DS18B20", "Address":"28FF1597A41604CE", "Temperature":25.81}}, "DHT22":{"Temperature":33.20, "Humidity":30.00}, "TempUnit":"C"}`

and this is the data from an other Wemos D1 mini with only a DHT22
`{"Time":"2017-10-05T22:39:55", "DHT22":{"Temperature":25.40, "Humidity":45.00}, "TempUnit":"C"}`

Is there any way of adding a unique identifier to all messages send by the same board?

davidelang, 2017-10-05:
do you mean that different wemos devices need to be identified in some way, or 
different DHT22 sensors connected to the same wemos device need to be 
identified?

jhlweb, 2017-10-05:
Yes every different Wemos devices needs to have some common ID in all its telemetry.

The DHT22 could then be easy identified because the pin number is attached when connecting multiple DHT22 sensor to the same Wemos device.

davidelang, 2017-10-06:
On Thu, 5 Oct 2017, jhlweb wrote:

> Yes every different Wemos devices needs to have some common ID in all its telemetry.

it's in the message, just not the json piece. Every message sent includes the 
name of the machine sending it. You are just throwing away that info before the 
part that you quoted.

ionciubotaru, 2017-10-06:
Set a different `topic` for each wemos device

jhlweb, 2017-10-06:
@davidelang its that so? How can I get this info? Its nowhere in my mosquitto log and also other webbased MQTT client do not show this data part.

davidelang, 2017-10-06:
every mqtt tool that I have seen or used has two parts to the message

first the topic, then the payload

what you posted is just the payload, the topic is something like 
stat/device/sensor for the sensor reports.

without the topic the payload is impossible to make sense of (the payload is 
normally something like On/Off, 0/1, etc

jhlweb, 2017-10-06:
@davidelang I posted the topic to. Its the standard project topic tele/sonoff/SENSOR  I use this for all board because on the server side I use a script that translate all messages, If I use a topic per board i need to listen al all those topics and translate them all separately.  

I think that the hostname would be a good choice for an ID to comeback in each json

davidelang, 2017-10-06:
On Fri, 6 Oct 2017, jhlweb wrote:

> @davidelang I posted the topic to. Its the standard project topic tele/sonoff/SENSOR  I use this for all board because on the server side I use a script that translate all messages, If I use a topic [...]

if you have multiple devices using the name 'sonoff' you are going to have lots 
of problems. Each device should be named differently. mqtt includes the wildcard 
# that you can listen to that will report everything.

stefanbode, 2017-10-07:
@jhlweb I totally agree to @davidelang. It makes no sense, that everything is reported in the same topic and you just parse the JSON to get an idea where it comes from. One example why it does not make sense is the LWT topic. (Last Will Topic) here you can see, if your device is still connected or not. If all devices report into the same this is a mess because this message is "retained". You can create a wildcard subscribe and then parse the topic AND the message. This should solve your problem to let everything running into one queue.

jhlweb, 2017-10-12:
@davidelang @stefanbode  Yes you're right in the standard Sonoff mode of Tasmota firmware this ID is not needed and with topic wild-cards I can still pass all SENSOR items to 1 variable in the code. 
But if, as I'm doing with 20 WEMOS D1 mini boards, you're building a home sensor array with OpenHab, MQTT, Grafana / Chronograf , InfluxDB, Telegraf and MYSQL it's not always possible to pass the MQTT topic with the data. At that moment data with no unique address, such as the DHT22 information and the INFO1,2,3 and STATE topics become useless. Yes this may also be an issue for the OpenHab GIT but adding a ID/Hostname to the STATE and SENSOR topic would be a nice feature. 
For sending commands you need to address all devices individually and adding a ID to those replies is useless.  
I added this code to both `state_mqttPresent` functions and in the headder i added `#define USE_ID` just like it's done with ADC_VCC.
`#ifdef USE_ID
  snprintf_P(mqtt_data, sizeof(mqtt_data), PSTR("%s, \"" D_CMND_HOSTNAME "\":%s"), mqtt_data, Hostname);
#endif`

davidelang, 2017-10-12:
what about all the mqtt messages that aren't JSON, they don't even tell you what 
button is being pressed, or what relay you are getting notified turned on, 
that's all in the topic.

If you are exporting this to stuff that needs it in JSON, then you need to add 
the topic to the message you are exporting rather than require that every 
message, and all other oftware change to using json and include the topic in the 
message.

jhlweb, 2017-10-14:
@davidelang That was what I said. It's a failure of OpenHab that it can't push the topic through. And I can't find a solution within OH and the issue is also killed there. So i'll add it myself to tasmota firmware every time i'm updating.

closed 2018-05-07 by stale[bot]