arendst/Tasmota #977: Connecting external LED

ElDuderinos, 2017-10-06:
Hi,
I've read a lot of the wiki and documentation but I could not understand whether it is possible, any help would be appreciated.
I want to connect an external led to one of the Gpios and make it behave just like the built-in Led. This way I want proper indication to the sonoff state. I see that there is a possibility to set the gpio as led - GPIO_LED1 for example.I couldn't understand what command do I need to send in order to light up the led. Also, is there a way to make it behave just like the built-in led instead of sending extra commands to the sonoff from my openhab?
If there is no proper way of doing it, I might just solder another led in parallel to the built-in one. Is this possible? To use 2 less in parallel? (I can cut off the built-in led but I'd rather have them both)
Thanks!

arendst, 2017-10-06:
Commands to control Led1(I) are ``LedState`` and LedPower``. Other leds are not supported at the moment.

You could change the default GPIO used by Led1I to the GPIO you are connecting your Led too. This way the inbuilt led is not used but your external led does perform as the inbuilt one.

Snipercaine, 2017-11-12:
Is it possible to have the onboard LED turn ON when the relay is in the OFF position? 
I've got mine in light switches with buttons and would like the switch to light up when the lights are off so they are easy to find in the dark. 

btw: Thanks for making this firmware Theo!

closed 2018-05-07 by stale[bot]