arendst/Tasmota #978: sonoff basic Inching /Self-Locking

marcgrc, 2017-10-06:
Hello to all.
Does anyone know if the sonoff basic with firmware tasmota  can be timed?
for example, is it only active until I press the button or
turns on after two seconds it goes off?
thank you

davidelang, 2017-10-06:
see the pulsetime command on the wiki commands page

marcgrc, 2017-10-06:
wher is wiki commands page ??

marcgrc, 2017-10-06:
Sorry 
Where is wiki commands page??

marcgrc, 2017-10-06:
ok found
Take an example of what to change to get a 2 second impulse and then unplug it
please
Thank you very mutch

davidelang, 2017-10-06:
to to the tasmota github repository, and at the top of the page there is a link 
to the wiki (one of the tabs), along the right side of the wiki are links to the 
most significant pages, one of those is the commands that can be understood by 
tasmota.

closed 2018-05-07 by stale[bot]

vdhub, 2019-02-02:
Hello, I just got an Sonoff SV and I need same as marcgrc said, I need it to act like temporary push button, I start it and after 2 seconds it switches off, ITEAD have this in their app but I want to use it outside the app. 

How can I do that. ?

Thanks

ascillato, 2019-02-02:
Yes, in Tasmota console use command pulsetime