arendst/Tasmota #981: Sonoff B1 change color and dim with domoticz

sth313, 2017-10-06:
I got my B1 working with tasmota, with the console the color could be changed.

Is it also possible to add the bulb to domoticz and change the color and dim the lights?

For now i can only turn it on and off with a RGBW dimmer selected.

arendst, 2017-10-08:
As I am unable to find any configurable and useable RGB configuration in Domoticz using MQTT please provide more information about Domoticz RGB posibilities for non zigbee devices.

sth313, 2017-10-09:
I didn't got it working with Domoticz either so I switched to Home Assistant with Hass.io. With HA i was able to use this configuration:

`light:
  platform: mqtt
  name: "Lounge lamp"
  state_topic: "stat/sonoff/POWER"
  state_value_template: "{{ value_json.state }}"
  command_topic: "cmnd/sonoff/POWER"
  rgb_state_topic: "stat/sonoff/COLOR/RESULT"
  rgb_command_topic: "cmnd/sonoff/COLOR"
  rgb_command_template: "{{ '#%02x%02x%02x%02x%02x' | format(red, green, blue, 00, 00)}}"
  brightness_state_topic: "stat/sonoff/RESULT"
  brightness_command_topic: "cmnd/sonoff/DIMMER"
  brightness_scale: 100    
  brightness_value_template: "{{ value_json.brightness }}"
  color_temp_state_topic: "stat/sonoff/RESULT"
  color_temp_command_topic: "cmnd/sonoff/CT"
  color_temp_value_template: "{{ value_json.color_temp }}"
  qos: 0
  payload_on: "ON"
  payload_off: "OFF"
  optimistic: false
  retain: true`

found on [link](https://community.home-assistant.io/t/how-to-inexpensive-10-us-wifi-rgb-bulb-that-works-with-home-assistant/14735/351) 

Meaby the code can be rewriten for domoticz?

closed 2017-10-11 by sth313

PeggyFree, 2017-10-11:
Leaving domoticz is a painful workaround. Rgbw management by MQTT is still
partial at domoticz.

HobbytronicsPK, 2019-03-24:
Use RGB WW Virtual Switch in Domoticz for Colour / Warm / Cold / Brightness etc. features