arendst/Tasmota #983: Tasmota  AS3935 based lightning and storm sensor module

jeylites, 2017-10-06:
Hi all 

is it possible to implement AS3935 lightning and storm sensor module with Tasmota? If not how can this be achieved using an Arduino board to send values & alert to Tasmota?

[image]
[image]

lgnd33, 2017-10-06:
Wow, this is nice :) Didn't know this existed :) too bad I don't have a
use-case for it at all...

Anwyay, seems these modules can use both SPI (which doesn't work with
Tasmota) as well as I2C (which is supported by Tasmota).
Tasmota can't read specific AS3935 data, but support for it can be added by
someone that has time + knowledge + actual module.

If you'd like to give it a go, I'd point you to one of the xsns "drivers"
that use I2C (for example xsns_sht1x.ino) , as well as their invocations in
sonoff.ino. Given that libraries exist, it shouldn't be too hard to put it
together.

Good luck.

closed 2018-05-07 by stale[bot]

ludwich66, 2020-04-05:
There is a Tasmota Version with AS3935 on Test now.

jeylites, 2020-04-05:
That's great!

jeylites, 2020-04-06:
@ludwich66 

is this on development or available on current version?

ludwich66, 2020-04-06:
Hi jeylites,

i am not the proger - i am the tester ;-)
Its a very good collaboration Projekt.
device111 progs without sensor - iam owner off a sensor (mod1016) without Programming skills

Please test it! - but report your Experience!!
GUI
MQTT output
what device do you use?
Praise and criticism?

Value	Description
AS3935default	-	
load default for Sensor and Settings (no argument required)
AS3935settings	-	show all settings no argument
AS3935calibrate	-	auto calibrate the internal Capacitors
AS3935setnf	0,1,2,3,4,5,6,7	Noise Floor Level Wert von 0-7 (3 Bit)
AS3935autonf	0,1	
Set Auto Tune for Noise Level 0/1 (1 Bit)
AS3935setgain	Indoors, Outdoors	Set Inddoors/Outdoors
AS3935autonfmax	0,1	
Set Auto Tune with Indoors and Outdoors 0/1 (1 Bit)
AS3935setwdth	0-15	
Watchdog threshold, value from 0-15 (4Bits)
AS3935settunecaps	0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15	
Internal Tuning Capacitor (from 0 to 120pF in steps of 8pf)
AS3935setrej	0-15	Spike rejection / Ablehnen
AS3935setml	1,5,9,16	Minimum number of Lightnings (2 bit)
AS3935disturber 	0,1	Set Disturber on/off ( 1 Bit)
AS3935autodisturber	0,1	
Set Auto-Disturber.On for 120sec. per Event
AS3935mqttevent	0,1	
json messages only for lightning events 0/1 (1 Bit)

ludwich

jeylites, 2020-04-06:
@ludwich66 

That's great; I will check it out once I get a chance to get back to my home town. I purchased the AS3935  modules on Alixepress, not too long ago, but never had an opportunity to implement it. This, however, would allow me to do so. 

I will update once I get things going.

ludwich66, 2020-04-06:
Great - please	 test it!
Today we are dealing with the finetuning.
its difficult to find a good setting between sensivite an disturber…
There is no real Stormy Weather...

if forgot to tell: there is a GPIO Setting „AS3935 (202)“
Now Its German based on tasmota 8.0.2
Plaese use AS395calibrate to find the correct setting for the Resonator

ludwich