arendst/Tasmota #987: 3 Pin Rocker Switch with LED

jtroberts1, 2017-10-07:
I have a 3 Pin rocker style switch, there is also a LED built into the switch.  What I am trying to do is use the switch as a simple ON/OFF control of a string of ws2812 led's but also illuminate the LED on the rocker to show status.  I am using a wemos D1 mini

setup like so....
WS2812 on GPIO13 = working great
Rocker switch on GPIO2  = (pin setup = Gold pin to GND, vcc to 5V, and middle pin to GPIO2.

The LED on the rocker illuminates on and off with the movement of the switch however the ws2812 lights do not turn on/off.

Now if I wire rocker to not use the built in led of the switch it turns the ws2812 ON/OFF but the built in LED does no illuminate.  Interesting enough the blue LED on the Wemos toggles with the rocker.

Any help is appreciated.  Thanks

davidelang, 2017-10-08:
how have you configured GPIO2 in software?

what do you have configured in your home automation software? you will probably 
need to have it detect the sitch position and control the WS2812 strip.

> Now if I wire rocker to not use the built in led of the switch it turns the ws2812 ON/OFF but the built in LED does no illuminate.  Interesting enough the blue LED on the Wemos toggles with the rocker.

it's not clear what you mean by this. what is the wiring here?

jtroberts1, 2017-10-08:
@davidelang , thanks for the response.

GPIO2 is configured as switch1.
GPIO13 is configured as WS2812
Switchmode1=0

I wasn't trying to involve HomeAssistant yet.  I was hoping I could do all of this directly within Tasmota.  Ultimately HA will also be able to turn off/on and control the ws2812 strip.

Here's what I want to do...
Rocker Swtich with built in status LED.  When Ws2812 lights are on Rocker LED should be on and vice/versa.  This rocker switch will also turn on/off the Ws2812 LED strip.

What I was trying to explain is if I just try to use the rocker as a typical SPST switch it works as expected.  IE: GND connected to GPIO2.  In that scenario the LED pin on the rocker is NOT connected but it DOES control the ws2812 strip.

for reference here is a [link](https://www.sparkfun.com/products/11155) to the rocker switch.

Any help is greatly appreciated.  Thanks

davidelang, 2017-10-08:
I haven't used WS2812 LEDs yet, but I though that they needed something more 
than On/Off, they are individually controlled, so you need a pattern that will 
be sent to the strip.

Switch1 will control Rely1 by default, it won't do anything with other things.

What is the command you use via MQTT to control the WS2812 LEDs?

jtroberts1, 2017-10-08:
They're all documented.  Color HEX.
Scheme 1,2,3 etc...  just execute from Tasmota console.

You need a logic level shifter but that's it.  Real simple.   The ws2812
isnt my issues.

davidelang, 2017-10-08:
my point is that the output of switch1 isn't color hex, it's on/off

by default switch1 will control relay1

ws2812 leds aren't relay1, so switch1 won't change them without the intervention 
of your home automation software.

jtroberts1, 2017-10-08:
Yes. I know this.  Please forget about the ws2812 and or any relay.  this
is NOT the issue or concern.  Also, remember I am not using a sonoff.  I'm
using a Wemos.

Also, let me clarify...  We have 2 (two) "LED's" in use...

1. Ws2812.  again, not relevant, just fyi about what I want to turn
on/off.  Let's not worry about how to change pattern or color, I have that
covered.

2. The Single LED built into the Rocker Switch.

So let me simplify the question....

I want to use a rocker switch with a built in LED to turn on/off something,
how can Tasmota, turn on/off "X", and use the built in LED of the rocker to
show the power status.

*Alternate solution....*   Again No home automation software.  all controls
done within Tasmota and the MCU

1. Ws2812 (GPIO13)
2. 2 Pin Rocker switch.  (GND, GPIO0) to control power.
3. Single Axial LED (GPIO5)

How to...
- When Rocker turns OFF the WS2812, the LED connected to GPIO5, is
illuminated
- When Rocker turns ON the WS2812, the LED connected to GPIO5, is NOT
Illuminated

This is all real simple but I just want physical controls and not have to
depend on HA software.

SO my experience is the LED is controlled by the PWM command.  So I guess
the question is how can Tasmota send a command to it's self.  IE PWM 500

Thanks

davidelang, 2017-10-08:
On Sun, 8 Oct 2017, jtroberts1 wrote:

> I want to use a rocker switch with a built in LED to turn on/off something,
> how can Tasmota, turn on/off "X", and use the built in LED of the rocker to
> show the power status.

the two have very little to do with each other. The Tasmota software has no way 
of controlling the LED in the switch (unless it has it's own input wire that 
accepts 3.3v)

If you hard-wire the led/switch so that the led changes with the switch 
position, and it does what you want, you can then configure the GPIO the switch 
is connected to as switch1 and (assuming you have a relay1), it will then 
control relay1. you can set switchmode to control what happens as the swich is 
changed, and you can set switchtopic to have it send a mqtt message instead of 
controlling relay1

without home automation software, you can't have the switch control anything 
else other than relay1.

> *Alternate solution....*   Again No home automation software.  all controls
> done within Tasmota and the MCU
>
> 1. Ws2812 (GPIO13)
> 2. 2 Pin Rocker switch.  (GND, GPIO0) to control power.
> 3. Single [...]

tasmota cannot send a command to itself, it either takes direct action for a 
relay, or it sends a message to the mqtt server. It has no way to do anything 
other than turn a relay on or off without external commands.

jtroberts1, 2017-10-09:
@davidelang,  thank you so much for you insight.  Of course I was hoping this wasn't the case but at least I can stop driving down the road of having Tasmota action on button presses etc...

I've started tinkering with the switchtopic option and using HA to "sense" the button press and then perform the specific action.  This should work well.  Thanks again.

closed 2017-10-09 by jtroberts1