arendst/Tasmota #988: DHT22 on 5.8.0 not working

mr-const, 2017-10-07:
I found out, that on clean installation of a 5.8.0 (and possibly 5.7.1) version, DHT22 sensor doesn't work, or works rarely (perhaps once in 5-10 minutes). Debug log contains `Timeout waiting for start signal low pulse` only.
However after rollback to 5.7.0, `reset 2`, reconfiguration, everything works as expected.

Upgrade from 5.7.0 to 5.8.0 works as expected, DHT22 works.
Tested on Wemos D1 mini and Sonoff Basic.

stefanbode, 2017-10-10:
I have the same issue right now and try to find out what is happening. with weblog 5 I did not see any message related to what you have reported. For me it did not work currently. reset 2 cause a Blocked Loop from webconsole. Therefore I have to figure out what is wrong.

ritschard, 2017-10-29:
Same issue. Finally found this issue after few days of thinking I'm going 🔩 😆

arendst, 2017-10-29:
See #1082

arendst, 2017-10-30:
Implemented in master release v5.9.0

closed 2017-10-30 by arendst