arendst/Tasmota #992: Add correction factors for R G and B for WS2812 NeoPixel strip

MikePRoberts, 2017-10-08:
Theo
First what you have done here is outstanding. Thank you and well done.
I have spent today playing with a WS2812B NeoPixel LED strip. When driving from a colour wheel I felt the Green was too strong and the Blue much too strong. I am driving the system with Node Red on a Pi. So I added the ability to apply a reduction to any of R G and B. I ended up correcting Green by 192/256 and Blue by 144/256. i.e, -25% and -44% respectively. (I also had LedTable =1) I tried to capture the improvement on video at: https://youtu.be/_sD2RtwXDHE The LEDs are a bit bright for the camera so please look around them for a better representation of what I am seeing. My suggestion would be a correction number for each of R G and B scaled to 256. So 256 is 100% and lower values reduce the intensity. The overhead is then a single byte multiply and >>8 for each colour value.
Thanks again.
Mike Roberts

arendst, 2017-10-08:
Mike,

does enabling ledtable (gamma correction) make for a better experience? Just enable it with command ``ledtable 1``.

MikePRoberts, 2017-10-08:
Theo

As noted I had already enabled LedTable.  I see that this applies the same correction to R G and B and hence does not correct for poor matching of the R G and B LEDs.  One would have hoped for a better balance in an integrated unit!  Also key areas where colours were a poor match were where two colours are at max brightness – where the LedTable has no impact.  Hence my action to scale G and B down to get a better fit.  I expect that LedTable is helping where the colours are less saturated.  I tried briefly with LedTable disabled but the correlation was definitely worse.

Mike

From: Theo Arends 
Sent: Sunday, October 8, 2017 9:53 AM
To: arendst/Sonoff-Tasmota 
Cc: Mike Roberts ; Author 
Subject: Re: [arendst/Sonoff-Tasmota] Add correction factors for R G and B for WS2812 NeoPixel strip (#992)

Mike,

does enabling ledtable (gamma correction) make for a better experience? Just enable it with command ledtable 1.

arendst, 2017-10-08:
Regarding scaling. The AiLight bulb has 1 Green, 1 Blue and 2 Red leds just to crank the Red up as you noticed.

Will think about implementation as it adds another item to the dimmer calculation ...

MikePRoberts, 2017-10-08:
Theo

Thank you for considering this.  Note in terms of calculation I envisioned applying the correction as the last step before sending the RRGGBB data to the LED string.  So all the other sums need not change.  This seems the right place to put it as we are correcting for LED intensity.

A further question – Would it help your quest for reducing program size to replace some floating point calculations with integer – or is the impact small?  I noticed there a few floating sums in the fade calculations.  Most of my programming experience is with PICs using assembly code where I have always found a fixed base/integer alternative.  If you think there is a worthwhile saving I will try to understand the code and see what would be involved.

A quick search suggests that the ESP8266 has a hardware 16 bit multiply.

Mike

From: Theo Arends 
Sent: Sunday, October 8, 2017 3:02 PM
To: arendst/Sonoff-Tasmota 
Cc: Mike Roberts ; Author 
Subject: Re: [arendst/Sonoff-Tasmota] Add correction factors for R G and B for WS2812 NeoPixel strip (#992)

Regarding scaling. The AiLight bulb has 1 Green, 1 Blue and 2 Red leds just to crank the Red up as you noticed.

Will think about implementation as it adds another item to the dimmer calculation ...

closed 2018-05-07 by stale[bot]