arendst/Tasmota #993: IRRemote stopped working after update to 5.8.0

bluebird11, 2017-10-08:
It used to work before, now I get:

 11:07:11 CMD: IRsend {"protocol":"NEC" ,"bits":32 ,"data":1286815935}
 11:07:11 MQT: stat/sonoff-livingroom-light/RESULT = {"IRSend":"NoPROTOCOL, BITS or DATA"}

No matter what command I send.

I have IRremoteESP8266 1.2.0 installed, the code doesn't compile with higher versions.

Did something change?

arendst, 2017-10-08:
Yes, something did change.

As a workaround please use uppercase words for PROTOCOL, BITS and DATA and report back if that solves the issue.

bluebird11, 2017-10-08:
Yes, that fixed it, thanks a lot and thank you very much for your work and the updates, SetOption13 is awesome!

Can you also consider adding support for 5 relays (main relay + 4 GPIO)? :)

arendst, 2017-10-30:
Implemented in master release v5.9.0

closed 2017-10-30 by arendst