arendst/Tasmota #996: Define gatewayserver as default timeserver

stefanbode, 2017-10-08:
I know this can be done by user_config_override, but I assume it is also a good idea to bring this into the general code as a default to use the gateway as timeserver. Nowadays most routers support actual time and can answer quite quick without name resolution and a request into the internet.

ionciubotaru, 2017-10-08:
Check the wiki:

NtpServer<x> |   | Show NTP server 1 to 3 name or ip address
NtpServer<x> | 0 | Set NTP server 1 to 3 name to none and restart
NtpServer<x> | 1 | Reset NTP server 1 to 3 name to user_config.h (NTP_SERVERx) and restart
NtpServer<x> | <ntphost> | Set NTP server 1 to 3 name or ip address (32 chars max)

stefanbode, 2017-10-10:
hi @ionciubotaru yes I'm aware of this. The router IP is normally changing and came through the DHCP configuration. Therefore "NtpServer1 default" with an internal reroute to the gateway would be a great improvement to avoid that all devices are calling the nameservers. Also, response time is better and even works, if the device normally has no internet access.

ionciubotaru, 2017-10-11:
The router WAN's IP is changing, while the router LAN's IP is static as long as the router act as a DHCP server (if not change it to static).

stefanbode, 2017-10-11:
@ionciubotaru. Sorry you did not get it to the point. For my server it is always 192.168.178.1, but other servers have other subnets. Therefore dynamically using this as first name server will solve it for all users. Tasmota knows the gateway and can easily substitute, if nameserver name is DEFAULT.

davidelang, 2017-10-11:
if you want your gateway to be your NTP server, define that in your DHCP 
settings. don't hard-code that assumption in the firmware code.

It may be reasonable to try that as a fallback if no NTP server is provided as 
part of the DHCP handshake, but don't override something if it is provided.

stefanbode, 2017-10-11:
This is also a good idea, to use the provided NTP server from DHCP as first choice. But even this need a  change in the firmware. Currently Tasmota ignores this and try NTP1

arendst, 2017-10-11:
Setting NTP server 0 to the default Gateway could be useful but...

I've been fiddling around with this but stumbled across two problems:

1 - For some reason I'm unable to typecast my Gateway Address configured as ``uint32_t sysCfg.address[1]`` into a ``ip_addr_t`` type needed for SDK ``sntp_setserver``.

2 - My default Gateway does not provide NTP. I'v tested that with the example from the SDK user guide and this code excerpt from support.ino from line 1115:
```
  ip_addr_t *addr = (ip_addr_t *)malloc(sizeof(ip_addr_t));
  ipaddr_aton("192.168.2.254", addr);
  sntp_setserver(0, addr);

  sntp_stop();
  sntp_set_timezone(0);      // UTC time
  sntp_init();

  free(addr);
```
Where the ip address shown is my Gateway. If I change the address to my local NTP server (192.168.2.27) it resolves times as expected. Ergo my Gateway is not a NTP server.

So the use case for me is none but I would like to know how to get the typecast going...

closed 2018-05-07 by stale[bot]