arendst/Tasmota #997: Change Ws2812 Color with a Button.  No HA Software

jtroberts1, 2017-10-08:
Let's say I have a Wemos.  I have the following items connected to individual GPIO's....

1. WS2812 strip
2. Button1
3. Button2
4. Button3

Ok, so if I open the Tasmota Console I can send a simple command to change the color of the WS2812.  "Color FF0000"  If I do this the LED's of the WS2812 change to red. Works perfectly and incredibly Simple.  Thanks Theo

So what I want to do is use the buttons to send the color command.  This way I have physical switches to change the color.  So let's lay it out...

When a user presses a button The color command is issued as follows and the Ws2812 changes color (RED, Green, Blue - IE:Button,1,2,3)

Button1 = Color FF0000
Button2 = Color 00FF00
Button3 = Color 0000FF

So basically the real question is how to get Tasmota to perform an action based on a button press.  We all know how to control power this way but how can we customize Tasmota to perform other commands when a button is pressed?

Thanks

davidelang, 2017-10-08:
On Sun, 8 Oct 2017, jtroberts1 wrote:

> Let's say I have a Wemos.  I have the following items connected to individual GPIO's....
>
> 1. WS2812 strip
> 2. Button1
> 3. Button2
> 4. Button3
>
> Ok, so if I open the Tasmota Console I can send a simple [...]

Tasmota cannot do this. the output of a switch is an on/off/toggle action of 
a local relay or an on/off/toggle command to amqtt server.

> When a user presses a button The color command is issued as follows and the Ws2812 changes color (RED, Green, Blue - IE:Button,1,2,3)
>
> Button1 = Color FF0000
> Button2 = Color 00FF00
> Button3 = Color [...]

change the source code.

David Lang

jtroberts1, 2017-10-08:
Can you give me some direction...  What file, and maybe line numbers to look at.  I'm not a developer, but I might be able to figure it out.  thanks

davidelang, 2017-10-08:
On Sun, 8 Oct 2017, jtroberts1 wrote:

> Can you give me some direction...  What file, and maybe line numbers to look at.  I'm not a developer, but I might be able to figure it out.  thanks

probably in sonoff.ino, I don't expect it to be an easy fix, what you are trying 
to do is a pretty significant change to how things work.

arendst, 2017-10-08:
Build a device with three buttons and let it send MQTT commands to your Tasmota loaded WS2812 connected device.

This way you do not have to rewrite Tasmota (as adding switches will be a major rewrite  of Tasmota as it is being a MQTT listener).

The device could be a Wemos D1 mini with your own code. Nice exercise to get into coding and MQTT. Use Tasmota as a starting point (Just use the basic stuff in sonoff.ino).

jtroberts1, 2017-10-24:
Thanks.  I modified my idea a little to simply add a status led to a switch (think an LED mounted through a project box).  I used switchtopic2 "ledstatus"  then in Home Assistant created an MQTT light to turn on/off the ledstatus when this topic is received.  Added to that a couple automation rules in home assistant to turn the ledstatus on or off depending on the power state of the switch.

I know it's "simple" but it's these little things that help me understand how to use Tasmota and HA to create fun projects.  Thanks again

closed 2017-10-24 by jtroberts1