arendst/Tasmota #1000: Feature Request: Advanced LED API

ageorgios, 2017-10-10:
Thanks for the great work!

It would be really great if we could have an advanced LED API like the Yeelight for example:
http://led/cm?cmnd=Color FF0000&effect=smooth&duration=1000

This would bring the Sonoff Tasmota for LEDs on a whole other level since many effects and atmospheres can be implemented.

stefanbode, 2017-10-10:
Have you checked the new option (5.3.0) to send: "Backlog". This allows putting several commands in row (max 16) and executes them with a predefined delay. This should allow the same as you have in ONE of the http requests above. https://github.com/arendst/Sonoff-Tasmota/wiki/Commands

ageorgios, 2017-10-10:
I didn't know of this thanks!

closed 2017-11-30 by ageorgios